nRF24/RF24 #468: writeAckPayload Problems

alessandroAmedei, 2018-12-05:


alessandroAmedei, 2018-12-06:
I am trying to send back in ack payload 32 bytes of data.
Using 250KBPS it doesn't work and  the Transmitter.write always returns FALSE.
Using 2MBPS it quite works, the Transmitter.write returns something like 5 times TRUE and 1 time FALSE.
How should i do to fix this?
The main code is:

TX
`boolean ack=false;
  if(radio.write(radioBuf, sizeof(radioBuf))){
    ack=true;
    if(radio.isAckPayloadAvailable()){
      Serial.println("DoneACK");
      radio.read(&ackFileName,sizeof(ackFileName));
      Serial.println(ackFileName);
    }
    else
      Serial.println("done NOT ACK"); 
    }
  else
  Serial.println("ACK FAIL");`

RX
`  if (radio.available()) {
    radio.writeAckPayload(1,&ack,sizeof(ack));
    radio.read(&text, sizeof(text));`

SETUP

`radio.begin();
  radio.setDataRate(RF24_2MBPS);
  radio.setPALevel(RF24_PA_MAX);

  
  radio.setAutoAck(true);

  radio.enableAckPayload();
  radio.enableDynamicPayloads();`

TMRh20, 2018-12-09:
You may need to adjust the retry delay. Radio.setRetries(15,15); May enable it to work at 250kbps

closed 2019-05-10 by TMRh20