nRF24/RF24 #470: the driver stuck at this loop while connection error or module not responding

Edward-CNRG-NTU, 2018-12-21:
https://github.com/nRF24/RF24/blob/a2b35d6594160263a5a8c460a9754dd71fba6d95/RF24.cpp#L820

hello,
first of all, many thanks for this powerful driver, I found it really useful.

Now I encounter an issue, i'm using this package on raspi 3 b+, hardware spi_dev driving method. (should be the default driving method when I installed this driver with the shell script, right?)

I program my application with python (thus I'm mainly using the python API). My application started out working smoothly. But after I deployed the program to the field, from time to time, the program got stuck in the loop where the driver kept printing "RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc." I thought, well may be the connection was loose somehow (I still using the jumper wires to connect the SPI bus). I re-did the connection and ran the program again. The program work fine again. Sometime this problem came up again, I visually inspected the connections (I didn't touch the wires, so assumed the wires were as it were) , nothing was loose.  (So maybe the triggering factor was the module not responding timeout.) And I restart the program (not even reboot the raspi), the program worked fine again. So I came up with an idea that the program was able to reinitialize the nrf24 module as a recovery plan when encountering that error again. (This will recover from the timeout error but definitely not the wiring error.)

**Long story short**, the problem was the driver stuck at this timeout/not-responding loop and never got a chance to get out and reinitialized the connection. (This is just my first glance at this code, so maybe the driver is clever enough to do the recovery and I didn't realize that.) I'm suggesting adding a break after printing the error maybe?

Many thanks

Edward Chen

TMRh20, 2018-12-21:
If you edit the RF24_config.h and uncomment #define FAILURE_HANDLING it will break the loop and set failureDetected = true;

TMRh20, 2019-05-10:
#494

closed 2019-05-10 by TMRh20