nRF24/RF24 #471: Problem with multiple transmitters [3xNRF24l01]

ZgnDaniel, 2018-12-28:
Hello ! 
I am totally new to GitHub so I am sorry if this is not the place to post this issue.

I want to control an Neopixel LED (16 led ring) with [UNO] when a person is detected (in front of a garage) with a motion sensor (PIR) [1st NANO] , then the light will turn from green to red on the neopixel, but this will only work if an contact is closed (garage door open) on the [2nd NANO].

Communication is made with **3x NRF24l01**
2 transmitters [NANO] and 1 receiver [UNO]

I worked in C and arduino just 2 days and now I am stuck and I am in a hurry :) I tried my best but I cannot see how it's done properly. I also have to make an **if** condition that will not enter if the door is open and in this condition will be the part where the LED wil light up.
I have codes that works individually but not all 3 together because I can't send and receive the infos correctly and condition is not finished yet. **[attached on a folder]**

If you can help me this will mean a lot for me. Thank you.

[Codes_ok.zip](https://github.com/nRF24/RF24/files/2715646/Codes_ok.zip)

Last errors I got are "'pipe' was not declared in this scope" for transmitters and expected initializer before 'RF24' for receiver

**Sensor Transmitter [NANO 1**]

```
//Include needed Libraries at beginning
#include "nRF24L01.h" //NRF24L01 library
#include "RF24.h"
#include "SPI.h"

#define sensorPin 2 // Sensor switch is connected to Pin 8 on NANO
int etatSensor[1] = {000}; // Used to store value before being sent through the NRF24L01

RF24 radio(9,10); // NRF24L01 used SPI pins + Pin 9 and 10 on the NANO

const uint64_t pipes[3] = { 0xF0F0F0F0E1LL, 0xF0F0F0F0E2LL, 0xF0F0F0F0E3LL }; // Needs to be the same for communicating between 2 NRF24L01 

int buttonState = 0; // etat de la sortie du capteur
const int ledPin = 4; // la LED du Arduino

void setup(void){
{
  pinMode(ledPin, OUTPUT); //la broche de la LED est mise en sortie
  pinMode(sensorPin, INPUT); //la broche du capteur est mise en entree
}

pinMode(sensorPin, INPUT_PULLUP); // Define the arcade switch NANO pin as an Input using Internal Pullups
digitalWrite(sensorPin,HIGH); // Set Pin to HIGH at beginning

radio.begin(); // Start the NRF24L01
radio.openWritingPipe(pipe[2]); // Get NRF24L01 ready to transmit
radio.startListening()
}

void loop(void) 
{
  buttonState = digitalRead(sensorPin);//lecture du capteur
  if (buttonState == HIGH) //si quelquechose est detecte
  {
    digitalWrite(ledPin, HIGH); //on allume la LED
  }
  else //sinon
  {
    digitalWrite(ledPin, LOW); //on eteint la LED
  }
  
if (digitalRead(ledPin) == LOW){ // If Switch is Activated
etatSensor[0] = 111;
radio.write(etatSensor, 1); // Send value through NRF24L01
}
else {
etatSensor[0] = 000;
radio.write(etatSensor, 1);
}
}
```
**Contact transmitter [NANO 2]**

```
// NRF24L01 Module Tutorial - Code for Transmitter using Arduino NANO

//Include needed Libraries at beginning
#include "nRF24L01.h" //NRF24L01 library
#include "RF24.h"
#include "SPI.h"

#define SwitchPin 8 // Arcade switch is connected to Pin 8 on NANO

bool etatBtn[1] = {000}; // Used to store value before being sent through the NRF24L01
RF24 radio(9, 10); // NRF24L01 used SPI pins + Pin 9 and 10 on the NANO
const uint64_t pipes[3] = { 0xF0F0F0F0E1LL, 0xF0F0F0F0E2LL, 0xF0F0F0F0E3LL };// Needs to be the same for communicating between 2 NRF24L01

void setup(void) {

  pinMode(SwitchPin, INPUT_PULLUP); // Define the arcade switch NANO pin as an Input using Internal Pullups
  digitalWrite(SwitchPin, HIGH); // Set Pin to HIGH at beginning

  radio.begin(); // Start the NRF24L01
  radio.openWritingPipe(pipe[1]); // Get NRF24L01 ready to transmit
  radio.startListening()
}

void loop(void) {

  if (digitalRead(SwitchPin) == LOW) { // If Switch is Activated
    etatBtn[0] = 111;
    radio.write(etatBtn, 1); // Send value through NRF24L01
  }
  else {
    SentMessage[0] = 000;
    radio.write(&etatBtn, 1);
  }
}

```
**Receiver [UNO]**

```
//Include needed Libraries at beginning
#include "nRF24L01.h" // NRF24L01 library created by TMRh20 https://github.com/TMRh20/RF24
#include "RF24.h"
#include "SPI.h"
#include "FastLED.h" // FastLED library for WS2812 RGB Stick http://fastled.io/

#define NUM_LEDS 16 // Number of leds on stick
#define LED_PIN 8 // Digital In (DI) of RGB Stick connected to pin 8 of the UNO

CRGB leds[NUM_LEDS]; // FastLED Library Init

bool etatBtn, etatSensor
RF24 radio(9, 10); // NRF24L01 used SPI pins + Pin 9 and 10 on the UNO
const uint64_t pipes[3] = { 0xF0F0F0F0E1LL, 0xF0F0F0F0E2LL, 0xF0F0F0F0E3LL }; // Needs to be the same for communicating between 2 NRF24L01

void setup(void) {

  FastLED.addLeds<NEOPIXEL, LED_PIN>(leds, NUM_LEDS); // Setup FastLED Library
  FastLED.clear(); // Clear the RGB Stick LEDs

  // Light up starting LED's
  for (int x = 0; x != NUM_LEDS; x++) {
    leds[x] = CRGB::Red;
  }
  FastLED.setBrightness(50);
  FastLED.show();

  radio.begin(); // Start the NRF24L01
  radio.openReadingPipe(1, pipes[1]);// Get NRF24L01 ready to receive
  radio.openReadingPipe(2, pipes[2]);
  radio.startListening(); // Listen to see if information received

  pinMode(LED_PIN, OUTPUT); // Set RGB Stick UNO pin to an OUTPUT
}

void loop(void) {

  while (radio.available())
  {
    // radio.read(ReceivedMessage, 1); Read information from the NRF24L01

    radio.read(&etatBtn, 1);
    radio.read(&etatSensor, 1);

    if (etatBtn[0] == 111) // Indicates switch is pressed
    {
      if (etatSensor[0] == 111) // Indicates sensor detect
      {
        for (int x = 0; x != NUM_LEDS; x++)
        {
          leds[x] = CRGB::Green;
          FastLED.show();
        }
      }
      else
      {
        for (int x = 0; x != NUM_LEDS; x++)
        {
          leds[x] = CRGB::Red;
          FastLED.show();
        }
      }
      delay(10);
    }
```

TMRh20, 2018-12-28:
You are missing an “s” on pipes, and on the reciever you are missing a “;” on the line

ZgnDaniel, 2018-12-29:
> You are missing an “s” on pipes, and on the reciever you are missing a “;” on the line
> […](#)

Thank you for pointing this out. I did the corrections.

closed 2019-05-10 by TMRh20