nRF24/RF24 #475: Esp8266 crashes on transmit with version 1.3.1 but works with 1.3.0

pborkargithub, 2019-01-16:
I tried communication between esp8266 module 12E and arduino nano. I tried the examples which came with the library but the first one itself started to cause the esp8266 to crash on transmitting. 

then i tried code from [https://forum.arduino.cc/index.php?topic=444832.15](url) as below.

the module received properly but used to crash when i tried to transmit. however when i comment out 
wirelessSPI.stopListening();
the transmission works fine. 

Then i tried to use the earlier version which is 1.3.0. here everything works fine. I have even tried the first getting started example with no isues. 

Thanks 

Receiver Code:

`
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

const int pinCE = 2;
const int pinCSN = 15;
byte gotByte = 0;

RF24 wirelessSPI(pinCE, pinCSN);
const uint64_t pAddress = 0xB00B1E5000LL; 

void setup()   
{
  Serial.begin(9600);
  wirelessSPI.begin(); 
  wirelessSPI.setAutoAck(1);                 
  wirelessSPI.enableAckPayload();             
  wirelessSPI.setRetries(5,15);             
  wirelessSPI.openReadingPipe(1,pAddress);     
  wirelessSPI.startListening();             
  wirelessSPI.printDetails();   
}

void loop() 
{   
    while(wirelessSPI.available()){
     wirelessSPI.read( &gotByte, sizeof(gotByte) );
     Serial.print("Recieved packet number: ");
     Serial.println(gotByte);
     Serial.print("Size: ");
     Serial.println(sizeof(gotByte));
    }
   
  delay(100);   
}`

Transmitter Code:

`
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

const int pinCE = 2;
const int pinCSN = 15;
byte counter = 1;
bool done = false;
RF24 wirelessSPI(pinCE, pinCSN);
const uint64_t pAddress = 0xB00B1E5000LL;   

void setup() 
{
  Serial.begin(9600);
  wirelessSPI.begin();   
  wirelessSPI.setAutoAck(1);
  wirelessSPI.enableAckPayload();   
  wirelessSPI.setRetries(5,15);             
  wirelessSPI.openWritingPipe(pAddress);       
  wirelessSPI.stopListening();
  wirelessSPI.printDetails();             
}

void loop() 
{
   if (!wirelessSPI.write( &counter, 1 )){ 
       Serial.println("packet delivery failed");     
   }
   else {
       counter++;
   }
    delay(750);
}
`

dangilaus, 2019-05-07:
The problem is still present in version 1.3.2

TMRh20, 2019-05-10:
I think this is a problem with the ESP8266 software, since it seems to relate to this [line 779](https://github.com/nRF24/RF24/blob/56f9bf41bd34721308aacf15df607ef7f1fb559c/RF24.cpp#L779) in RF24.cpp where progmem is used. 

The line can be changed to ''' write_register(EN_RXADDR,read_register(EN_RXADDR) | _BV(ERX_P0)); ''' and it should work. If that is the case, maybe its a good idea to open an issue with ESP8266, since it should support progmem without crashing.

closed 2020-03-29 by TMRh20