nRF24/RF24 #483: Error compiling for digispark spi.h

electricplant, 2019-02-20:
Hey all, I'm about 10 problems deep trying to start at rock bottom. I've got a fresh install of the arduino ide trying to compile just the spi.h library in a blink sketch for the default digispark.

and I get this error: 

`Arduino: 1.8.8 (Windows 10), Board: "Digispark (Default - 16.5mhz)"

C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\hardware -hardware C:\Users\DGDYN\AppData\Local\Arduino15\packages -tools C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\tools-builder -tools C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\hardware\tools\avr -tools C:\Users\DGDYN\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\libraries -libraries C:\Users\DGDYN\Documents\Arduino\Arduino Sketchbook\libraries -fqbn=digistump:avr:digispark-tiny -vid-pid=0X2A03_0X0043 -ide-version=10808 -build-path C:\Users\DGDYN\AppData\Local\Temp\arduino_build_623140 -warnings=none -build-cache C:\Users\DGDYN\AppData\Local\Temp\arduino_cache_796205 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.avr-gcc.path=C:\Users\DGDYN\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\DGDYN\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -verbose C:\Users\DGDYN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\esp8266\examples\Blink\Blink.ino
C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\arduino-builder -compile -logger=machine -hardware C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\hardware -hardware C:\Users\DGDYN\AppData\Local\Arduino15\packages -tools C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\tools-builder -tools C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\hardware\tools\avr -tools C:\Users\DGDYN\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\arduino-1.8.8-windows\arduino-1.8.8\libraries -libraries C:\Users\DGDYN\Documents\Arduino\Arduino Sketchbook\libraries -fqbn=digistump:avr:digispark-tiny -vid-pid=0X2A03_0X0043 -ide-version=10808 -build-path C:\Users\DGDYN\AppData\Local\Temp\arduino_build_623140 -warnings=none -build-cache C:\Users\DGDYN\AppData\Local\Temp\arduino_cache_796205 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.avr-gcc.path=C:\Users\DGDYN\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\DGDYN\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -verbose C:\Users\DGDYN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\esp8266\examples\Blink\Blink.ino
Using board 'digispark-tiny' from platform in folder: C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7
Using core 'tiny' from platform in folder: C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7
Detecting libraries used...
"C:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10808 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "C:\\Users\\DGDYN\\AppData\\Local\\Temp\\arduino_build_623140\\sketch\\Blink.ino.cpp" -o nul
"C:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10808 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\SPI" "C:\\Users\\DGDYN\\AppData\\Local\\Temp\\arduino_build_623140\\sketch\\Blink.ino.cpp" -o nul
"C:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10808 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\SPI" "C:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\SPI\\SPI.cpp" -o nul
Generating function prototypes...
"C:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10808 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\SPI" "C:\\Users\\DGDYN\\AppData\\Local\\Temp\\arduino_build_623140\\sketch\\Blink.ino.cpp" -o "C:\\Users\\DGDYN\\AppData\\Local\\Temp\\arduino_build_623140\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files\\arduino-1.8.8-windows\\arduino-1.8.8\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\DGDYN\\AppData\\Local\\Temp\\arduino_build_623140\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10808 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\DGDYN\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\SPI" "C:\\Users\\DGDYN\\AppData\\Local\\Temp\\arduino_build_623140\\sketch\\Blink.ino.cpp" -o "C:\\Users\\DGDYN\\AppData\\Local\\Temp\\arduino_build_623140\\sketch\\Blink.ino.cpp.o"
In file included from c:\users\dgdyn\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5\avr\include\avr\io.h:99:0,

                 from c:\users\dgdyn\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5\avr\include\avr\interrupt.h:38,

                 from C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/WProgram.h:8,

                 from C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/Arduino.h:4,

                 from C:\Users\DGDYN\AppData\Local\Temp\arduino_build_623140\sketch\Blink.ino.cpp:1:

C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI/SPI.h: In member function 'void SPISettings::init_AlwaysInline(uint32_t, uint8_t, uint8_t)':

C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI/SPI.h:146:16: error: 'SPE' was not declared in this scope

[... cut ...]
C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI/SPI.h:310:55: error: 'SPIE' was not declared in this scope

   inline static void detachInterrupt() { SPCR &= ~_BV(SPIE); }

                                                       ^

Using library SPI at version 1.0 in folder: C:\Users\DGDYN\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI 
exit status 1
Error compiling for board Digispark (Default - 16.5mhz).`

If I didn't ask correctly or posted in the wrong area please let me know, github is very confusing but I can't find answers anywhere so I have to ask. Thank you

Avamander, 2019-02-22:
Your Arduino installation is broken, none of the errors you pasted here are in RF24 but are caused by your installation's SPI library.

closed 2019-02-22 by Avamander

electricplant, 2019-02-22:
> 
> 
> Your Arduino installation is broken, none of the errors you pasted here are in RF24 but are caused by your installation's SPI library.

yeah thats what I had assumed, so I reinstalled the ide but that didn't do the trick. I think I should manually reinstall it then. thanks