nRF24/RF24 #484: receiving wrong values added by arbitrary constant

yahyakandri, 2019-03-06:
here is my problem when i send data though rf24 i have a constant added  example:
i send data.i=100 i receive data.i =108 i use 2 nrf in 2 rpi 3 model b and the other rpi3 model b+
code receive
`#include <iostream>
#include <RF24/RF24.h>

RF24 radio(RPI_BPLUS_GPIO_J8_15,RPI_BPLUS_GPIO_J8_24, BCM2835_SPI_SPEED_16MHZ);
//RF24 radio(22, 0);
const uint8_t address[6] = "00001";
struct package
{
int i[5];
int y;
int z;
};
typedef struct package Package;
Package data;
void setup(void) {
radio.begin(); 
  radio.setRetries(15,15);
  radio.enableAckPayload();               // Allow optional ack payloads
  //radio.enableDynamicPayloads();
  // radio.enableDynamicPayloads();
  radio.setPayloadSize(32);
  radio.setChannel(90); 
  radio.setPALevel(RF24_PA_MAX);
  radio.setDataRate( RF24_2MBPS) ; 
    radio.openReadingPipe(1, address);
  radio.startListening();
  radio.printDetails();
}

int main(int argc, char** argv) {
  setup();

  while (true) {
     if ( radio.available()) 
  {std::cout<<"radio valide \n";
    while (radio.available())
    {
      radio.read( &data, sizeof(data));
      //if (data.i==3)
    std::cout<<sizeof(data)<<"\n";
    }
    
    std::cout<<data.i[0]<<"\n";
    std::cout<<data.i[1]<<"\n";
    std::cout<<data.i[2]<<"\n";
    std::cout<<"--------------------------------------------------------"<<"\n";
   
}

}
}
`
code send
 `#include <iostream>
#include <RF24/RF24.h>

RF24 radio(RPI_BPLUS_GPIO_J8_15,RPI_BPLUS_GPIO_J8_24,BCM2835_SPI_SPEED_8MHZ);
const uint8_t address[6] = "00001";
struct package
{
int i[5];
int y;
int z;
};
typedef struct package Package;
Package data;
void setup(void) {
radio.begin(); 
  radio.setRetries(15,15);
  radio.enableAckPayload();               // Allow optional ack payloads
  //radio.enableDynamicPayloads();
   //radio.enableDynamicPayloads();
  radio.setPayloadSize(32);
  radio.setChannel(90); 
  radio.setPALevel(RF24_PA_MIN);
  radio.setDataRate(RF24_2MBPS) ; 
  radio.openWritingPipe(address);
  radio.printDetails();
}

int main(int argc, char** argv) {
  setup();
  data.i[0]=1003;
  data.i[1]=500;
  data.i[2]=500;
  while(true){
radio.write(&data, sizeof(data));
  
   std::cout<<sizeof(data)<<"\n";
    std::cout<<data.i<<"\n";
   
}

}
`
if any could help me i would be grateful

Avamander, 2019-03-06:
Does the issue also exist with the examples that you haven't modified?

yahyakandri, 2019-03-07:
The ping-pong example do not work

Avamander, 2019-03-08:
That's not very specific, what does "do not work" mean?

yahyakandri, 2019-03-08:
in gettingstarted program 'failed, failed response time out '
event when i switch the roll

TMRh20, 2019-05-21:
This works fine for me
```
--------------------------------------------------------
radio valide
28
1003
500
500
--------------------------------------------------------
```

closed 2020-03-29 by TMRh20