nRF24/RF24 #485: label:"help wanted" pyRF24 problem with RP 3 B+

icemanx, 2019-03-08:
Hi,

Currently I am trying to run pingpair_dyn.py on Raspberry Pi 3 B+. I compiled libraries and I am able to run gettingstarted example. However, I'd like to use Python wrapper. I'm running the code with $sudo ./pingpair_dyn.py but program stucks at radio.begin(). No errors shown in the terminal.

I am using bcm2835 drivers. 

If you can guide me I'll be very glad. Thanks!

thebluetitan001, 2019-03-08:
Im having this same issue, id love to get some feedback as well

Avamander, 2019-03-08:
Can you confirm SPI actually works on your rPi?

icemanx, 2019-03-08:
Yes I can confirm SPI is working. I am tested the example gettingstarted.cpp between Raspberry and Arduino.

JakFourie, 2019-05-13:
This fixed my issues
In lib_nrf24.py in function:

def begin(self, csn_pin` ... after  self.spidev.open`add:

    self.spidev.max_speed_hz = 4000000

  [1]: https://raspberrypi.stackexchange.com/questions/71730/possible-spi-issues-around-nrf24-operations-post-rpi-system-update

homoroselaps, 2019-08-25:
I have the same issue right now. Have an Raspberry Pi 1 rev2 model B
the cpp example works with `sudo ./gettingstarted`
the `pingpair_dyn.py` example shows following output:
```
pyRF24/examples/pingpair_dyn/
Traceback (most recent call last):
  File "pingpair_dyn.py", line 67, in <module>
    radio.begin()
RuntimeError: unidentifiable C++ exception
```
Could you go into more detail @JakFourie how you managed to fix the problem? (There is no lib_nrf24.py in the repo)

closed 2020-03-29 by TMRh20