nRF24/RF24 #488: analogRead  issue

zwiebelslayer, 2019-04-17:
Hello
i try to send a struct of 3 floats an one int over nrf with the library.
 
Problem:
SendStruct.bodentemp=analogRead(A2);
and then sending the struct. i will either receive an 0 or an ohter random number

SendStruct.bodentemp=770;
works perfectly fine i will receive 770.

How can this be both values are ints.

I already have tried an array same results.

TMRh20, 2019-04-17:
What Arduino devices are you using? If 32 bit, make sure you declare your ints as uint16_t or int16_t for a 16-bit integer.

zwiebelslayer, 2019-04-17:
I am using two arduino nano.

TMRh20, 2019-04-17:
Hmm, can you post your code, or an example?

zwiebelslayer, 2019-04-17:
`#include <Arduino.h>
#include <Adafruit_BME280.h>
#include <Adafruit_Sensor.h> 
#include <DallasTemperature.h>
#include <OneWire.h>
#include <Wire.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
//_____________________________________________________________
#define SEALEVELPRESSURE_HPA (1013.25)
//_____________________________________________________________

#define onewirepin 4
OneWire oneWire(onewirepin);

DallasTemperature sensors(&oneWire);
DeviceAddress Probe = { 0x28, 0xFF, 0x7C, 0xD4, 0x60, 0x17, 0x05, 0x13 };

Adafruit_BME280 bme; // I2C
RF24 radio(7, 8); // CE, CSN
const byte address[6] = "00001";
typedef struct
{
  float LutftTemp;
  float LuftFeucht;
  float BodenTemp;
   int16_t    BodenFeucht;
} 
B_t;

B_t SendStruct;

float LuftTemperatur;
float LuftFeuchtigkeit;
float BodenTemperatur;

// Send Array______________

double  SendArray1[4];
int AnalogLesen;
int SendArray2[3];

//------------------------------------------=======================________________________
void DallasTemp(DeviceAddress devideadress)
{
SendArray1[3]=sensors.getTempC(devideadress);
delay(10);

SendStruct.BodenTemp = sensors.getTempC(devideadress);

}
void Messen1()
{
SendArray1[1]=bme.readTemperature();
SendArray1[2]=bme.readHumidity();

AnalogLesen=analogRead(A2);

      
sensors.requestTemperatures();
          DallasTemp(Probe);

          LuftTemperatur = bme.readTemperature();
           LuftFeuchtigkeit=bme.readHumidity();
  
        
        SendStruct.LutftTemp=bme.readTemperature();
        SendStruct.LuftFeucht = bme.readHumidity();
        AnalogLesen=map(AnalogLesen,0,1023,0,100 );

          SendStruct.BodenFeucht=analogRead(A2);

        radio.write(&SendStruct, sizeof(SendStruct));

}

void setup() {
  // BME STARUTP
  Serial.begin(9600);
   Serial.print("starte sendevorgang ");
   
   bme.begin(0x76);
   
   //BME ENDE
   //NRF START
    radio.begin();
  radio.openWritingPipe(address);
 radio.setPALevel(RF24_PA_MAX);
 radio.setRetries(15,15);
  radio.stopListening();
  
   
   
   // PINMODE//
 
 

}

void loop() {
 
  delay(200);
  Messen1();
Serial.println(SendStruct.LutftTemp);
      Serial.println(SendStruct.LuftFeucht);
          Serial.println(SendStruct.BodenTemp);
              Serial.println(SendStruct.BodenFeucht);
  
  delay(3000);
 
 

}`

`include <Wire.h>
#include <SPI.h>

#include <nRF24L01.h>

#include <RF24.h>

#include <Arduino.h>
#include <Wire.h>
int text;
int test =39;
int LuftTemp=19;
int Temp=10;
int ID;
int Ausgabe;
RF24 radio(7, 8); // CE, CSN
const byte address[6] = "00001";
float EmpfangsArray1[4];
float EmpfangsArray2[3];

typedef struct
{
  float LutftTemp;
  float LuftFeucht;
  float BodenTemp;
  int16_t BodenFeucht;
} B_t;

B_t EmpfangStruct;

float LuftTemperatur;
float LuftFeuchtigkeit;
float BodenTemperatur;
int BodenFeuchtigkeit;

int Eingang;

void requestEvent()
{
  Wire.write(text); // respond with message of 6 bytes
  
  
  // as expected by master
}

void receiveEvent(int howMany) {
  while (1 < Wire.available()) { // loop through all but the last
     ID = Wire.read();
    Serial.print(ID);
    Ausgabe=ID;

}
}
void setup()
{ 
   Serial.begin(9600);
    Serial.print("starte empfänger");
  Wire.begin(8);                // join i2c bus with address #8
  Wire.onRequest(requestEvent); // register event
  Wire.onReceive(receiveEvent);

  radio.begin();
  radio.openReadingPipe(0, address);
  radio.setPALevel(RF24_PA_MAX);

  radio.startListening();

  radio.setRetries(15, 15);
  pinMode(6, OUTPUT);
  
  
 

  

}

void loop()
{
  if (radio.available())
  {
    
    radio.read(&EmpfangStruct, sizeof(EmpfangStruct));
    Serial.println("Radioavailable");
 }
 
  delay(500);
  Serial.println(EmpfangStruct.LutftTemp);
      Serial.println(EmpfangStruct.LuftFeucht);
          Serial.println(EmpfangStruct.BodenTemp);
              Serial.println(EmpfangStruct.BodenFeucht);
}
`

zwiebelslayer, 2019-04-17:
Or easier to read
Sender: `typedef struct
{
  float LutftTemp;
  float LuftFeucht;
  float BodenTemp;
   int16_t    BodenFeucht;
} 
B_t;

B_t SendStruct;

SendStruct.LutftTemp=bme.readTemperature();
        SendStruct.LuftFeucht = bme.readHumidity();
        AnalogLesen=map(AnalogLesen,0,1023,0,100 );

          SendStruct.BodenFeucht=analogRead(A2);

        radio.write(&SendStruct, sizeof(SendStruct));

`

zwiebelslayer, 2019-04-17:
Also what does not work is;

AnalogSend=analogRead(A2);

radio.write(&AnalogSend, sizeof(AnalogSend));

TMRh20, 2019-04-17:
Testing your code, it works fine for me. Have you tried tying the analog
pin to GND or 5V to see if it remains stable at 0 or 1024?

TMRh20, 2019-04-17:
Also I would try printing your analog value on the sending side to see if
it is matching the recipient side. Just in case there is something funky
going on with the sender.

zwiebelslayer, 2019-04-17:
I print the whole structure out i reads for the analog value 789.
I am using a Capacitive Soil moisture sensor on this pin.
So when i connect A2 to 5V it read 1024 consistently.

TMRh20, 2019-04-17:
Weird, looks like your analog read is working fine. I’m stumped.

zwiebelslayer, 2019-04-17:
But i can not connect the Receiver and Sender to my PC at the same time i only have one of these weird USB cables. But i will connect an I2C Oled display to the sender that prints out the anlog Read Value.

zwiebelslayer, 2019-04-17:
OK thank you for your help when i connect my arduino nano via USB it works
when i am using external power my I2C display displays only 1 or 2 so not what it should read so it is definetly a wiring problem and has nothing to do with the code.

zwiebelslayer, 2019-04-17:
There was one wire wrong so it only had 3.3V but for the BME280 this was enough but for the moisture sensor not. Now i rewired it and it works perfect 
Thanks