nRF24/RF24 #491: ATTiny84 and NRF24L01 Can't get anything to work.

electricplant, 2019-04-20:
Hey all, I'm having a difficult time getting an ATTiny84 to work with an NRF24L01 in any way.
I know that my connections are good (to my knowledge) , I've added a diagram hopefully easily explaining the connections. 

I can guarantee that the radio modules themselves work perfectly because I've dedicated another proto board to testing them and there have been no issues. (An ESP8266 connected to the NRF sending to an Arduino with another NRF and it always receives/sends flawlessly. I then take the NRF from the ESP and hook it up to the ATTiny84) 

I've also posted in the Arduino forums [https://forum.arduino.cc/index.php?topic=611043.new#new](url) 
but I'm hoping someone here could help in any way

The code for the Arduino is this
``````
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
RF24 radio(7, 8); // CE, CSN
const byte generaladdress[5] = {1,'e','l','l','o'} ;
int moist = 5;
byte ackData[3] = {2,0,0};
byte datacommands[3] = {0,0,0};

void setup() {
  Serial.begin(9600);
  
  radio.begin();
radio.setDataRate( RF24_250KBPS );
  
  radio.openReadingPipe(1, generaladdress);
  radio.setPALevel(RF24_PA_MIN);
  radio.setAutoAck(true);
  radio.writeAckPayload(1, &ackData, sizeof(ackData));
  
  radio.startListening();
  radio.enableAckPayload();
  radio.enableDynamicAck();
 //radio.setAutoAck(false);
  
}
void loop() {
 
if (radio.available()){
    radio.read( &datacommands, sizeof(datacommands));  
      radio.writeAckPayload(1, &ackData, sizeof(ackData));
  Serial.print(datacommands[0]);
      Serial.print(" , ");
      Serial.print(datacommands[1]);
      Serial.print(" , ");
      Serial.println(datacommands[2]);
      Serial.println();
  }
 

}

``````

The ATTiny code is this

The blinker() is just my form of a serial response since I don't have a uart to usb thing

``````
#define __AVR_ATtiny84__ //not sure if this helps
#include <nRF24L01.h>
#include <RF24.h>
#include <EEPROM.h>
#include <math.h>

#define CE_PIN  2 //connected to physical pin 11
#define CSN_PIN 3 //connected to physical pin 10
RF24 radio(CE_PIN, CSN_PIN);

const byte generaladdress[5] = {1,'e','l','l','o'} ;
int i = 0;
int led = 0 ;

byte datacommands[3] = {0,0,0};
byte ackData[3] = {2,0,0};

void setup() {

pinMode(led,OUTPUT);
blinker();
radio.begin();
  radio.setDataRate( RF24_250KBPS );
  
blinker();
radio.openWritingPipe(generaladdress);
  radio.stopListening();

  radio.enableAckPayload();
  radio.enableDynamicAck();
  blinker();
}

void loop() {
  
for(i=0;i<3;i++) { 
  radio.powerUp(); // Found this in a tutorial. Tried with and without, not sure if it affects anything
  delay(5);
  radio.write( &datacommands, sizeof(datacommands)); 

blinker();

radio.powerDown(); // Found this in a tutorial. Tried with and without
  
  delay(1000);
}

}

void blinker() {
  digitalWrite(led,HIGH);
  delay(300);
  digitalWrite(led,LOW);
  delay(300);
}

``````
connections are here:
vcc and gnd as always
LED           --> Physical pin 2,     Arduino IDE 0
NRF CE      --> Physical pin 11,   Arduino IDE 2
NRF CSN    --> Physical pin 10,   Arduino IDE 3
NRF SCK    --> Physical pin 9
NRF MOSI  --> Physical pin 7
NRF MISO  --> Physical pin 8

[image]

If I'm making any sort of huge mistake I don't see, don't feel bad letting me know, I'm still new to this and don't claim I know 100% of what I'm doing. I've commented where I am unsure of the code. But the rest I've tested hundreds of times with arduino uno so I'm confident it should work.

If anyone has any tips or examples that worked for them I'd love to try it out. Ive done 5 different tutorials today and would just love to at least find one that works. This current code I've written myself from the basic examples included in the library.

dparson55, 2019-05-19:
Try switching the MISO and MOSI pins.  The radio's MISO should be connected to physical pin 7 on the ATtiny84.

[image]

[image]

electricplant, 2019-05-21:
> 
> 
> Try switching the MISO and MOSI pins. The radio's MISO should be connected to physical pin 7 on the ATtiny84.
> 
> [image]
> 
> [image]
> 
> Note the Arduino pin names (pictured in brown) are the values used with the [MIT High-Low Tech](http://highlowtech.org/?p=1695) ATtiny boards library https://github.com/damellis/attiny.

Hey, thanks for the idea, but sadly I already tried it, no luck there either. 
I'm going to look at the link you included in the bottom though, haven't seen that before, thanks

dparson55, 2019-05-26:
I tried out the code as well and no luck on my end either.  I got it working between 2 Arduino Uno's but not with an ATtiny84, running it at both 1MHz and 8MHz.  You can always give my radio library a try, it's called NRFLite.

dparson55, 2019-05-26:
I opened a pull request to hopefully make using the library with the ATtiny84 easier, but I noticed a far better improvement implemented in https://github.com/nRF24/RF24/pull/326.  It's over 2 years old though so maybe RF24 isn't the library to use for ATtiny's.

TMRh20, 2019-05-26:
Ok, ok I’ll take a closer look at that pr soon.

dparson55, 2019-05-26:
Having dived into this more I should have initially replied to this issue by saying to use the https://code.google.com/archive/p/arduino-tiny/ ATtiny core.

I'll be testing it today but it's fallen out of usage.  Everyone uses these 2 core and unfortunately they have different pin mappings than the core on Google Code.
https://github.com/damellis/attiny
https://github.com/SpenceKonde/ATTinyCore

TMRh20, 2019-05-26:
I just tested this with ATTiny85, the current code and the code from #326 and they both work with the SpenceKonde core available in the IDE. Pin mapping is per the diagram at https://tmrh20.github.io/RF24/ATTiny.html

dparson55, 2019-05-27:
Awesome and thanks!  Hopefully the OP will close this issue now. 

Interesting how the ATtiny84 pins got so out of order across the cores.  Made myself a picture for future reference:

Google Code Core (PA6 is Arduino 4)
```
// ATMEL ATTINY84 / ARDUINO
//
//                           +-\/-+
//                     VCC  1|    |14  GND
//             (D  0)  PB0  2|    |13  AREF (D 10)
//             (D  1)  PB1  3|    |12  PA1  (D  9) 
//                     PB3  4|    |11  PA2  (D  8) 
//  PWM  INT0  (D  2)  PB2  5|    |10  PA3  (D  7) 
//  PWM        (D  3)  PA7  6|    |9   PA4  (D  6) 
//  PWM        (D  4)  PA6  7|    |8   PA5  (D  5)        PWM
//                           +----+
```

SpenceKonde Core **not** reversed (PA6 is Arduino 4)
From https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/variants/tinyX4/pins_arduino.h#L157
```
// ATMEL ATTINY84 / ARDUINO
//
//                           +-\/-+
//                     VCC  1|    |14  GND
//             (D  0)  PB0  2|    |13  PA0  (D 10)        AREF
//             (D  1)  PB1  3|    |12  PA1  (D  9)
//             (D 11)  PB3  4|    |11  PA2  (D  8)
//  PWM  INT0  (D  2)  PB2  5|    |10  PA3  (D  7)
//  PWM        (D  3)  PA7  6|    |9   PA4  (D  6)
//  PWM        (D  4)  PA6  7|    |8   PA5  (D  5)        PWM
//                           +----+
```

SpenceKonde Core **reversed** (PA6 is Arduino 6)
From https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/variants/tinyX4_reverse/pins_arduino.h#L157
```
// ATMEL ATTINY84 / ARDUINO
//
//                           +-\/-+
//                     VCC  1|    |14  GND
//            (D  10)  PB0  2|    |13  PA0  (D  0)        AREF
//             (D  9)  PB1  3|    |12  PA1  (D  1)
//             (D 11)  PB3  4|    |11  PA2  (D  2)
//  PWM  INT0  (D  8)  PB2  5|    |10  PA3  (D  3)
//  PWM        (D  7)  PA7  6|    |9   PA4  (D  4)
//  PWM        (D  6)  PA6  7|    |8   PA5  (D  5)        PWM
//                           +----+
```

MIT Core (PA6 is Arduino 6)
From https://github.com/damellis/attiny/blob/master/variants/tiny14/pins_arduino.h#L33
```
// ATMEL ATTINY84 / ARDUINO
//
//                           +-\/-+
//                     VCC  1|    |14  GND
//             (D 10)  PB0  2|    |13  AREF (D  0)
//             (D  9)  PB1  3|    |12  PA1  (D  1) 
//                     PB3  4|    |11  PA2  (D  2) 
//  PWM  INT0  (D  8)  PB2  5|    |10  PA3  (D  3) 
//  PWM        (D  7)  PA7  6|    |9   PA4  (D  4) 
//  PWM        (D  6)  PA6  7|    |8   PA5  (D  5)        PWM
//                           +----+
```

electricplant, 2019-05-27:
> 
> 
> Finally got things working so hopefully you can too. It was an issue with pin numbers for the ATtiny board library I mentioned. So if you use the board library https://github.com/damellis/attiny, here's the necessary update.
> 
> **Change the USI Arduino pin numbers in [RF24.cpp around line 1603](https://github.com/nRF24/RF24/blob/master/RF24.cpp#L1603)**
> 
> ```
> #elif defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
> // Pin numbers for ATtiny board library https://github.com/damellis/attiny
> #	define DI   6   // PA6 - Physical pin 7 - USI Data Input pin - Connect to radio MISO
> #	define DO   5   // PA5 - Physical pin 8 - USI Data Output pin - Connect to radio MOSI
> #	define USCK 4   // PA4 - Physical pin 9 - USI Clock pin - Connect to radio SCK
> #	define SS   3   // (not used by USI)
> ```
> 
> **ATtiny84 TX**
> 
> ```
> #include <nRF24L01.h>
> #include <RF24.h>
> 
> // Pin numbers for ATtiny board library https://github.com/damellis/attiny
> #define LED_PIN 0 // PA0 - Physical pin 13
> #define CE_PIN  2 // PA2 - Physical pin 11
> #define CSN_PIN 3 // PA3 - Physical pin 10
> 
> RF24 radio(CE_PIN, CSN_PIN);
> const byte generaladdress[5] = { 1,'e','l','l','o' };
> byte datacommands[3] = { 0,0,0 }; 
> 
> void setup()
> {
>     pinMode(LED_PIN, OUTPUT);
> 
>     radio.begin();
>     radio.setDataRate(RF24_250KBPS);
>     radio.setPALevel(RF24_PA_MIN);
> 
>     radio.openWritingPipe(generaladdress);
>     radio.stopListening();
> }
> 
> void loop()
> {
>     for (int i = 0; i<3; i++)
>     {
>         datacommands[0]++; // Change data command so we can see it change on the RX side.
> 
>         if (radio.write(&datacommands, sizeof(datacommands)))
>         {
>             blink(1); // Blink 1 time for success.
>         }
>         else
>         {
>             blink(3); // Blink 3 times for failure.
>         }
> 
>         delay(1000);
>     }
> }
> 
> void blink(byte count)
> {
>     while (count--)
>     {
>         digitalWrite(LED_PIN, HIGH);
>         delay(100);
>         digitalWrite(LED_PIN, LOW);
>         delay(100);
>     }
> }
> ```
> 
> **Arduino Uno RX**
> 
> ```
> #include <nRF24L01.h>
> #include <RF24.h>
> 
> #define CE_PIN  9
> #define CSN_PIN 10
> 
> RF24 radio(CE_PIN, CSN_PIN);
> const byte generaladdress[5] = { 1,'e','l','l','o' };
> byte datacommands[3] = { 0,0,0 };
> 
> void setup()
> {
>     Serial.begin(115200);
>     
>     radio.begin();
>     radio.setDataRate(RF24_250KBPS);
> 
>     radio.openReadingPipe(1, generaladdress);
>     radio.startListening();
> }
> 
> void loop()
> {
>     if (radio.available())
>     {
>         radio.read(&datacommands, sizeof(datacommands));
> 
>         Serial.print(datacommands[0]);
>         Serial.print(" , ");
>         Serial.print(datacommands[1]);
>         Serial.print(" , ");
>         Serial.print(datacommands[2]);
>         Serial.println();
>     }
> }
> ```

Oh fantastic. I will test it out as soon as I get home today, thank you

heavyeste, 2020-02-05:
I can't find the code to modify in :
[https://github.com/nRF24/RF24/blob/master/RF24.cpp#L1603](https://github.com/nRF24/RF24/blob/master/RF24.cpp#L1603)

The library is changed and now I can find:

```c++
void SPIClass::begin() {
    // set USCK and DO for output
    // set DI for input
        #if defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__)
    DDRB |= (1 << PB2) | (1 << PB1);
    DDRB &= ~(1 << PB0);
        #elif defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
    DDRA |= (1 << PA4) | (1 << PA5);
    DDRA &= ~(1 << PA6);
        #elif defined(__AVR_ATtiny2313__) || defined(__AVR_ATtiny4313__)
    DDRB |= (1 << PB7) | (1 << PB6);
    DDRB &= ~(1 << PB5);
        #elif defined(__AVR_ATtiny861__)
    DDRB |= (1 << PB2) | (1 << PB1);
    DDRB &= ~(1 << PB0);
        #endif // defined(__AVR_ATtiny861__)
    USICR = _BV(USIWM0);
}
```

How can change this cose for the right pin?

Thank you

miker2069, 2020-03-29:
> I can't find the code to modify in :
> https://github.com/nRF24/RF24/blob/master/RF24.cpp#L1603
>
> The library is changed and now I can find:
>
> ```c++
> void SPIClass::begin() {
>     // set USCK and DO for [...]

I can't seem to find my detail reply on this topic but I believe the issue is you're editing RF24 library 1.3.3 or 1.3.4 - The mods don't work with that, you have to use 1.3.2.  I tried following a lot of the suggestions here but they didn't work. I ended up following EXACTLY the instructions in this [youtube video](https://www.youtube.com/watch?v=z7ejMTxf8rE).  

I used David M. Ellis's board manager and followed the video - now it all works perfectly.  The key is to make the mods to RF24 v1.3.2.

2bndy5, 2020-09-21:
ok to close this? It sounds like OP solved this on their end. Is this issue pending a required change to the RF24 library or the merging of a PR?

electricplant, 2020-09-22:
> ok to close this? It sounds like OP solved this on their end. Is this issue pending a required change to the RF24 library or the merging of a PR?

Hi, sorry I abandoned this project about a year ago. It was a project I started without enough knowledge of microcontrollers and my first github post so I had no clue what to do. Yes it's completely fine to close

2bndy5, 2020-09-22:
no worries

closed 2020-09-22 by 2bndy5