nRF24/RF24 #492: '-Wstrict-prototypes' is valid for C/ObjC but not for C++, Error compiling python

Diallomm, 2019-04-25:
running build
running build_ext
building 'RF24' extension
arm-linux-gnueabihf-gcc -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c pyRF24.cpp -o build/temp.linux-armv6l-3.5/pyRF24.o
cc1plus: warning: command line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++

Avamander, 2019-04-25:
It's a warning though, I heavily doubt it'd halt at a warning unless something is seriously wrong. Try waiting a bit longer for the files to compile.

Diallomm, 2019-04-25:
Ah! ok I did not understand this, but when I look at the process on Htop the processor is 100% and all the ram is consumed, suddenly the rpi block, I compile it on a raspberry pi v1 512mo

Diallomm, 2019-04-27:
"Build takes several minutes on arm-based machines. Machines with RAM <1GB may need to increase amount of swap for build. "
Oh! now i know why my compilation not work, i don't have increased my swap on my RPv1

closed 2019-05-10 by TMRh20