nRF24/RF24 #493: RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.

GordonTraveler, 2019-05-04:
Hi.

I know this issue is reportet already. I just have a new question to it. 

I have this problem from time to time. Everything is running like a charm for some (random) time... days, weeks and then - suddenly - I get this "RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc." error. The biggest problem is that it breaks the whole program around it. Even CTRL+C is not working anymore. Nothing. It is not possible to stop this loop.

Can anyone tell me how to 'convert' this endless error message loop into a normal 'Error handling'... in meaning of "end the try to send data with a error message in return". That would help me a lot! My software would continue working and I could deal with the error myself.

I know it is not solving the main problem. But it would help me already.

TMRh20, 2019-05-04:
See the rf24 docs at http://tmrh20.github.io/RF24/classRF24.html#a2e40fe66d1231a333aa2534e8491f828

With failure handling enabled, you should be able to bring the radio back online in many cases by calling radio.begin() and reconfiguring its settings.

GordonTraveler, 2019-05-05:
Thank you TMRh20. I will try it.

I guess a new radio.begin() will not help (but I will try). Usually if this error happens the whole nrf24l01+ is not responding at all anymore. There are 2 stages. First I just get a timeout problem. Short after I get this hardware fail error. In the stage of the timeout problem a reboot of the RPi is helping. But with the harware fail error I have to completely take the power of the RPi or pull the nrf24l01+ off and put it back again (it is on a socket, not soldered to a board). 

I have the system in an RV. So the search of the main problem is quite complicated. Especially that the failure occures absolutely random after quite a long time. Until now I could not reproduce it.
I can live with it... but the biggest problem is the hardware fail message loop... that breaks all the system, because it is blocking this thread (which will never end) and I don't get a return message, so I could stop the system to start new threads which try to send data via the nrf24l01+. So until I realize that the nrf24l01+ got hung up again, hundreds of never ending threads are running on the PRi. Not nice.

closed 2019-05-21 by TMRh20