nRF24/RF24 #494: Enable failure handling by default?

TMRh20, 2019-05-08:
Just looking for input on whether to enable the config variable "failure_handling" in RF24_config.h by default. It adds a bit of extra code to the library, but seems necessary to deal with random radio errors that seem to be failures with the radio module itself.

Basically the driver will get locked in an endless loop if something goes wrong with the radio, and it becomes unresponsive. Failure handling adds timeouts to the write functions to prevent and detect this type of error.

I've found that over random periods of time, my RPi (RF24Mesh) master node will encounter this error and freeze up, and have had "failure handling" enabled on my devices for some time now. 
The thing is, some of my Arduino devices do it, and some are rock solid...

Are others experiencing these random errors? I've got two RPi devices, and it happens randomly on both of them. 

I just fixed a loop in RF24Network that also would lock up in certain failure modes, and added a timeout to correct that as well, so with the latest code and failure handling enabled, radio modules are hot-swappable, with recovery from all errors.

Thanks for any input on this.

TMRh20

TMRh20, 2019-05-17:
Wow, never expected this to be such a hot topic...

So all in all there seem to be three types of errors these modules will encounter. It seems best to have a single function that checks for ``` radio.failureDetected == true ``` and re-configures the radio as per the following:

1. Writing to radio: Radio unresponsive - Fixed internally by adding a timeout to the write functions in RF24 (failure handling)
2. Reading from radio: Available returns true always - Fixed by adding a timeout to available functions by the user. example:

```
uint32_t timer = millis();
while radio.available(){
if(millis() - timer > 1000){ radio.failureDetected = true; break; }
radio.read etc...
}
```

3. Radio configuration settings are lost - Fixed by monitoring a value that is different from the default, and re-configuring the radio if this setting reverts to the default.

```
if(radio.getDataRate() != RF24_1MBPS){
  radio.failureDetected = true;
}
```

If a failure is detected via any of these methods, calling radio.begin() and re-configuring settings appears to fix the problem. I'll add this to the documentation when I add failure handling by default.

Avamander, 2019-05-17:
I am indifferent about enabling failure handling, both defaults work but maybe indeed failure handling by-default would be better for newcomers. If the change is done then the examples should be updated as well, to show which failures the library can handle.

TMRh20, 2019-05-17:
Done, added an example and updated the documentation. Will likely update higher layer examples at a later date.

closed 2019-05-17 by TMRh20