nRF24/RF24 #500: Write returns false, even on successful transmit

benjaminvdb, 2019-06-09:
I'm sending data from an Arduino Uno to a Raspberry Pi using NRF24L01+ modules. Things are working out quite well, but I found something odd that I cannot explain.

Here is a snippet from my code:

```
char message[] = "Hello world!";
for (int i = 0; i < MAX_RETRIES; i++) {
    if (!radio.write( &message, sizeof(message) )){
        Serial.println(F("ERROR: could not transmit message. Check wires."));
        continue;  // retry
    }
    break;
}
```

The thing is that `write` often returns `false` here, while I can see that the message is received on the Raspberry Pi. This happens often, but not always. What could be the problem here?

TMRh20, 2019-06-09:
It’s the nature of wireless. You can adjust the max retries and intervals to maximize responses, but you are just not getting an ack response from the RPi sometimes.

benjaminvdb, 2019-06-09:
Thanks a lot for the quick response! I think I understand. I'm very new to both Arduino and dealing with wireless in this way, so I wasn't sure what went wrong. In the documentation for `write()` I read the following:

> [returns] True if the payload was delivered successfully false if not

Perhaps it would be more clear if this stated something like:

> True if an ACK response was received from the receiving end, indicating the payload was delivered successfully

This also better defines the behavior of `write()` in case auto-acknowlede is disabled (or at least this is what I would expect, though I haven't tested it). Again, I'm a total noob, so feel free to correct me if I'm wrong.

Thanks a lot for maintaining this library, it has been of great help!

TMRh20, 2019-06-09:
Makes sense, I’ll have to update that in the docs.

benjaminvdb, 2019-06-09:
Thanks again, @TMRh20! 😃

closed 2019-06-09 by benjaminvdb