nRF24/RF24 #501: Implementing to stm32 HAL

EminAtes, 2019-06-09:
Hi,

    I have attempt to make compitable this library with stm32 hal library, currently i can communicate to nrf24l01+ chip. However, i am not able to communicate between two nodes(stm32 based nodes). I shared screenshots related to issue. 

I have tested my hardware with arduino nanos(instead of stm32s to be sure) and it is working fine. Also I have setup spi with 2 stm32 board directly and it is working fine. SPI periperal clocks under 10 Mhz.
Receive examples screenshot: [Receive](https://ibb.co/546v8Dz)
For testing receive example I set up arduino based transmitter node which transmits single data every 50ms (I am sure it is transmitting ). The screenshot dont show data but it is always same data and the value is "14". But arduino based transmitter sends 0x41(A).

Do you have suggestions for debugging etc.?

Avamander, 2019-06-10:
First thing I'd suggest is compare variable sizes. I suspect basic data type sizes are different, we've had this with ESP8266 and ESP32, if you find a place where a specific size is needed please do open a PR that uses `uint[size]_t` instead (or some other fix is required).

EminAtes, 2019-06-11:
Finally, I was able to access an oscilloscope and compare SPI signals between arduino based nrf24l01l+ node and the stm32 based nrf24l01+ one. There are only 2 thing different between each other. One of them voltage level (of course it should be different) and other one is MOSI lines default state. Stm32 based node have LOW level MOSI and Arduino based node have HIGH level MOSI line. But I think it shouldnt be problem. Because slave spi devices figure out data signal with clock signal. Even so, I will add extenal pull up resistor to stm32 mcu and will share results. I hope it works...

EminAtes, 2019-06-12:
I've been working on it for 2-3 days, but I didn't make it work. Can you take a look SPI and clock [configuration](https://github.com/EminAtes/NRF24/blob/master/Src/main.cpp) please ? You can review the code, If you want. I uploaded all project.

Pluscrafter, 2019-06-12:
I would have interest to look at the code, beacuse I would also have an inplemention of this libarary to STM32 for my project.

EminAtes, 2019-06-12:
Sorry my bad, I set it as private repository by mistake... I change it to public. Can you try again please ?

Pluscrafter, 2019-06-12:
It works. I will look at this on saturday.

EminAtes, 2019-06-12:
By the way Do you have working implementation?

Pluscrafter, 2019-06-12:
No this is why I want to help.

Pluscrafter, 2019-06-12:
I had also problems with SPI when I want to communicate with an ICM-20689 beacuse the NSS line gone High in the half of the transaction.

EminAtes, 2019-06-12:
I have good news ! I set timeout value as 0xffffff and it works while receiving. Hovewer It didnt works as transmitter...

Pluscrafter, 2019-06-12:
A little big for timeout.

Pluscrafter, 2019-06-12:
try to add 
while (HAL_SPI_GetState(&hspix) != HAL_SPI_STATE_READY);
before endTransaction();
maybe the transaction isn't ended before the next start.

EminAtes, 2019-06-12:
Actually I tried a few different value and result didnt change. By the way I don't understand how works timeout. However, As you mention the problem is timing. So I will work on timing as soon as possible when i access an oscilloscope again. Thanks a lot for your help !

Pluscrafter, 2019-06-12:
I will try to implement this also with DMA to reduce the load of the microcontroller. On the STM32 SPI timing is really a little bit hard to code. I would try to probe the NSS(csn) and SCK line and look if the NSS line goes to fast high like in the picture below.
[image]

This is a picture of a sucsessful transaction.
[image]

EminAtes, 2019-06-12:
I look for how to deal with nss by software and i found [this post](https://community.st.com/s/question/0D50X00009XkbYmSAJ/spi-nss-configuration-and-use-help-please). You can read directly last answer. I will try it tomorrow.

Pluscrafter, 2019-06-12:
Will see if this work.
`uint8_t rcv;
GPIOx->BSRR = (1<<r_Bit);
HAL_SPI_Transmit_DMA(&spi,&msg,1);
HAL_SPI_Receive_DMA(&spi,&rcv,1);
while (HAL_SPI_GetState(&spi) != HAL_SPI_STATE_READY); 
GPIOx->BSRR = (1<<s_Bit);

return rcv;`

This line I get from
`while (HAL_SPI_GetState(&spi) != HAL_SPI_STATE_READY); `
I get from
https://github.com/fboris/STM32Cube_FW_F4/blob/master/Projects/STM32F4-Discovery/Examples/SPI/SPI_FullDuplex_ComDMA/Src/main.c

This is my implementation for a TransmitReceive with DMA. Try this if the example above isn't working. 

`GPIOx->BSRR = (1<<s_Bit);`
is HAL_GPIO_WRITE. s_Bit for 1 and r_Bit for 0

Pluscrafter, 2019-06-13:
```
uint8_t RF24::write_register(uint8_t reg, const uint8_t* buf, uint8_t len)
{
	uint8_t status;
	uint8_t temp = (W_REGISTER | ( REGISTER_MASK & reg));
	beginTransaction();
	HAL_SPI_TransmitReceive(hspix, &temp, &status, 1, (0));
	HAL_SPI_Receive(hspix, (uint8_t *) buf, len, (0));
	endTransaction();

	return status;
}
```
I think you want to write to the registers not read from it. This might be the problem you can't transmitt any data.

EminAtes, 2019-06-17:
You are right ! I made mistake there. I accidentally delete some file from project repository while undo commit. I will make changes on library and share it. Thank you for your help.

Pluscrafter, 2019-06-21:
Could you commit the missing files, that i can create an dma access also.

EminAtes, 2019-06-21:
I didn't readjust the library yet. I am designing&drawing custom board(i didnt design pcb antenna before so it takes some time). I have planned to finish it this weekend and I will order pcbs than I will start to implement library again because I didnt have copy of project on my local. I would inform you when i finished the project.

raghur, 2019-06-24:
Possibly check this - https://github.com/Hvunt/nrf24l01p

EminAtes, 2019-06-27:
@raghur the aim is this project is not only using nrf24l01. I want to use also network topology , mesh and etc. That is why I am tring to implement this library to stm32 hal. I manage to implment the library but i accidentally delete project. I am going to spent all this weekend to reconstruct the project and i will commit  it..

EminAtes, 2019-06-29:
@Pluscrafter, I finally got to where I left off. Now I am working on transmiting data issue. Receiving data working well as before... 
I create clean repository for [the project](https://github.com/EminAtes/NRF24_STM32).

EminAtes, 2019-06-30:
@Pluscrafter, Did you review the project ? The problem of not being able to transmit data still persists. Do you have any suggestion ?

EminAtes, 2019-06-30:
@Pluscrafter you can add dma access to the project. I start to test mesh and network libraries...

closed 2019-06-30 by EminAtes