nRF24/RF24 #507: how to build python liberary?

imohammad74, 2019-07-11:
Hello,
Thanks for your library,I finally after 2 weeks can connected Arduino to RPi 3 b+. I follow [your doc](http://tmrh20.github.io/RF24/Python.html) ,But when try ``` ./setup.py build``` , i get this error:
```python
Traceback (most recent call last):
  File "./setup.py", line 39, in <module>
    process_configparams()
  File "./setup.py", line 19, in process_configparams
    identifier, value = line.split('=', 1)
ValueError: need more than 1 value to unpack
 ```  
Could you please help me fixed it?

architeuthisss, 2019-09-06:
I think your problem is because Makefile references Makefile.inc, so changing Makefile to Makefile.inc makes Makefile reference itself.

michz, 2019-10-19:
Hi @imohammad74 ,
For me it helped to build the library itself before trying to build the python wrapper:

```bash
git clone https://github.com/nRF24/RF24.git
cd RF24.git
./configure
make
sudo make install
cd pyRF24
./setup.py build
sudo ./setup.py install
```

The doc is probably missing a note that you have to build the library itself before.

closed 2020-01-28 by Avamander