nRF24/RF24 #508: ESP32 and disabled SPI transactions

Mariusz1, 2019-07-17:
I develop a project with the ESP32, Nrf24L01 and the SPI ILI9341 display. Everything works fine except that the display was extremely slow. This happend after executing RF24 methods interacting with the radio. After some investigations I found out that the SPI transactions are not enabled in the RF24 library.

In the _"RF24_config.h"_ file, there is the line 33:
`#if defined (SPI_HAS_TRANSACTION) && !defined (SPI_UART) && !defined (SOFTSPI)`
It looks like that for some reason the `SPI_HAS_TRANSACTION` is false. This is strange because it is correctly defined in the ESP32 file (line 25) _"esp32-hal-spi.h"_ included via the _"Arduino.h"_.

As a fix for the problem I have changed (by adding `|| defined (ESP32)`) the above line (_"RF24_config.h"_ file) to:
`
#if (defined (SPI_HAS_TRANSACTION) || defined (ESP32))  && !defined (SPI_UART) && !defined (SOFTSPI)
`
Maybe, somebody more skilled will fix that in a better way?

I also propose to add a utility method for testing the SPI transaction support:

- in the _public_ section of the _"RF24.h"_ file 
`/** Indicates if the SPI transactions are supported in the current environment (board). */
	bool isSpiTransactionSupported();`
- in the _"RF24.cpp"_ file: 
`
    bool RF24::isSpiTransactionSupported()
    {
    #if defined (RF24_SPI_TRANSACTIONS)
       return true;
    #endif    
    return false;
    }
`

The problem is probably also related to the _SPI Configuration on Teensy_ #477.
If it matters, I use the PlatformIO.

Avamander, 2019-07-19:
Are you sure you're using that version of ESP32 core that actually has `SPI_HAS_TRANSACTION`?

Avamander, 2019-07-19:
Another possibility is that SPI_UART or SOFTSPI being defined, can you confirm those aren't defined?

Mariusz1, 2019-07-20:
> Another possibility is that SPI_UART or SOFTSPI being defined, can you confirm those aren't defined?

This is not the case because my fix solves the problem (and the "defined" are connected with the "&&"):
`
#if (defined (SPI_HAS_TRANSACTION) || defined (ESP32))  && !defined (SPI_UART) && !defined (SOFTSPI)
`

> Are you sure you're using that version of ESP32 core that actually has `SPI_HAS_TRANSACTION`?

I have the following entries in the _"platformio.ini"_ file:

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
```
If I use the VSCode "go to definition" it points to the _"....platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.h"_ file which has the following line:
`#define SPI_HAS_TRANSACTION`

Please, feel free to request other checks/information.

Avamander, 2019-07-20:
Can you change the `#define SPI_HAS_TRANSACTION`in that file to `#define SPI_HAS_TRANSACTION 1` and see if that fixes the issue without your fix applied?

This seems very weird but it might be yet another minor implementation detail I've forgotten :P

Mariusz1, 2019-07-20:
> Can you change the `#define SPI_HAS_TRANSACTION`in that file to `#define SPI_HAS_TRANSACTION 1` and see if that fixes the issue without your fix applied?
> 
> This seems very weird but it might be yet another minor implementation detail I've forgotten :P

Unfortunately the new `define` did not fix the problem.

Avamander, 2019-07-20:
If you add
```
#ifdef SPI_HAS_TRANSACTION
  #error "SPI_HAS_TRANSACTION TEST"
#endif
```
Does the compiler error out or compile the sketch nicely?

Mariusz1, 2019-07-20:
Yes, there is an error during compilation.
If I write `#ifndef SPI_HAS_TRANSACTION` then there is no error (as expected).

Avamander, 2019-07-20:
But if you put that code into `RF24_config.h` then it doesn't error out?

Mariusz1, 2019-07-20:
That's really weird. I tried a few times and both statements produce an error:
`#ifndef SPI_HAS_TRANSACTION`
`#ifdef SPI_HAS_TRANSACTION`
The error is dissmised only when I remove the statements. I do not know how it is possible :(

Avamander, 2019-07-20:
Does `#if defined(SPI_HAS_TRANSACTION)` error as well?

Mariusz1, 2019-07-20:
Yes. And the following code emits an error, too
```
#if !defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 2"
#endif
```

Avamander, 2019-07-20:
That is seriously weird... 
So what you're saying
```
#if defined(SPI_HAS_TRANSACTION) && !defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif
```
Errors out and displays that message?

Mariusz1, 2019-07-20:
I tried something else (just to test the directives). It is OK (no error):
```
#if defined(__AVR_ATxmega64D3__)
  #error "SPI_HAS_TRANSACTION TEST 2"
#endif
```
But the following code produces an error (as expected because the ATxmega is not defined):
```
#if !defined(__AVR_ATxmega64D3__)
  #error "SPI_HAS_TRANSACTION TEST 2"
#endif
```

Mariusz1, 2019-07-20:
> That is seriously weird...
> So what you're saying
> 
> ```
> #if defined(SPI_HAS_TRANSACTION) && !defined(SPI_HAS_TRANSACTION)
>   #error "SPI_HAS_TRANSACTION TEST 3"
> #endif
> ```
> 
> Errors out and displays that message?

No error.

Avamander, 2019-07-20:
Try this again and make sure you save the file you edited just to make sure this isn't a human error, does this error out in RF24_config.h?

```
#if defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION is defined"
#endif
```

Mariusz1, 2019-07-20:
No error:
```
#if defined(SPI_HAS_TRANSACTION) && !defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif
```

Error:
```
#if defined(SPI_HAS_TRANSACTION) 
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif
```

Error:
```
#if !defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif
```

Avamander, 2019-07-20:
Make sure you put the lines after `#define __RF24_CONFIG_H__` in `RF24_config.h`.

I have the suspicion that you have some other library doing `#undef SPI_HAS_TRANSACTION` and that library should be fixed instead.

Mariusz1, 2019-07-20:
Error:
```
#if defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION is defined"
#endif
```

```
Compiling .pio\build\esp32doit-devkit-v1\src\MyRadio24.cpp.o
In file included from src\MyRadio24.cpp:2:0:
C:/Users/Mariusz/.platformio/lib/RF24_ID433/RF24_config.h:45:4: error: #error "SPI_HAS_TRANSACTION is defined"
   #error "SPI_HAS_TRANSACTION is defined"
    ^
```

Mariusz1, 2019-07-20:
> Make sure you put the lines after `#define __RF24_CONFIG_H__` in `RF24_config.h`.

I put the code after the line:
```
#if defined (SPI_HAS_TRANSACTION) && !defined (SPI_UART) && !defined (SOFTSPI)
  #define RF24_SPI_TRANSACTIONS
#endif 
```
> I have the suspicion that you have some other library doing `#undef SPI_HAS_TRANSACTION` and that library should be fixed instead.

That was my guess, too.
I will try to investigate it.

Mariusz1, 2019-07-20:
I performed a text search for `SPI_HAS_TRANSACTION` in all installed libraries, but there are no `undef` directives relaated to the `SPI_HAS_TRANSACTION`. Wierd...

Avamander, 2019-07-20:
So just to sum it up:
  * No library does `#undef SPI_HAS_TRANSACTION`
  * Erroring when SPI_HAS_TRANSACTION is defined works as expected
  * Erroring when SPI_HAS_TRANSACTION isn't defined also works??

Have I understood you correctly?

Can you try adding `#error "RF24_SPI_TRANSACTIONS"` as line 33 in RF24_config.h, does it error out?

Avamander, 2019-07-20:
Oh, one more thing, have you made sure you're on the latest version of RF24?

Mariusz1, 2019-07-20:
> Oh, one more thing, have you made sure you're on the latest version of RF24?

Yes, I use 1.3.3.

> Can you try adding `#error "RF24_SPI_TRANSACTIONS"` as line 33 in RF24_config.h, does it error out?

Yes, there is an error.

Maybe, the order of includes shoud be different? In my cpp file I have:
#include <SPI.h>
#include <RF24_config.h>
#include <RF24.h>
#include <printf.h>
#include <nRF24L01.h>

#include "MyRadio24.h"

Mariusz1, 2019-07-20:
In my cpp file, inside my method:
No error:
```
#if !defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif
```
Error:
```
#if defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif
```

In the "RF24_config.h" file:
Error:
```
#if defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif
```
Error:
```
#if !defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif
```
However, the above code is grayed (inactive) in the VSCode IDE, but compiler generates an error.

Avamander, 2019-07-30:
Okay looking back at this now, I think it's some kind of dev environment issue. Can you put
```#if defined(SPI_HAS_TRANSACTION)
  #error "SPI_HAS_TRANSACTION TEST 3"
#endif

Mariusz1, 2019-07-30:
> In `platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.h`, the only reasonable explanation is that something else gets used as the board core.

Different locations in the _esp32-hal-spi.h_ file:

- the test `#if` before `#define SPI_HAS_TRANSACTION`: no error
- the test `#if` after `#define SPI_HAS_TRANSACTION`: error

Avamander, 2019-12-15:
Sorry, I forgot to reply. The result means something that is included later somehow undefines it. If you have grep try finding `#undef SPI_HAS_TRANSACTION` in your environment please.

Mariusz1, 2019-12-15:
I do not have grep, but I tried to find in files. No exact match, but below there are result of the "#undef" in my global library cache (vscode/Platform IO).
[image]

Avamander, 2019-12-16:
I'll ask the ESP32 core maintainers

sedwano, 2020-02-09:
Hi guys. I've had the same problem as Mariusz1. I started following the advice by Avamander, but compilation crashed when I tried to enable SPI_transactions. At best the display would be very slow. I gave up. the display works fine before the radio is activated. I tried turning all power to RF24 off when the display would update but no improvement,  and modified the RF24 library as suggested by nhatuan84 for soft SPI  using the HSPI pins. the RF24 still worked when running alone. but failed when the display was activated on the VSPI. I also added SPI.end() functions to both display (ILI9488) and RF24 libraries, no luck. 
Then it was time for random experimentation. and by luck the first thing I tried worked :-)

The active line for ESP32 (MH32-mini devkit) in the SPI clock divider is for 
           #elif F_CPU < 320000000
	     _SPI.setClockDivider(SPI_CLOCK_DIV2); // was DIV32. Test with DIV2
I changed it to DIV2 instead of DIV32 and suddenly everything worked like a charm. 
This is from line54 of RF24.cpp

Im running speed checks on the RF communication because I need to communicate with a battery operated display that does a lot of heavy GPS and other calculations that take program memory, so I really wanted the RF24 to work, as it is both faster and much lighter than both WiFi, BLE and BT classic. It just wasnt acceptable with the slow display. Now I can proceed with RF24 :-)

Sorry for not showing the code properly. I intend to read up on github protocoll soon. I just wanted to share. 

Regards sedwano.

Avamander, 2020-02-09:
The nRF24L01+ maximum SPI speed is 10Mbps, with a 80MHz or even a 40MHz clock a division by 2 means a bus way too fast for the nRF. How can it work? 

Some ESP32 expert should really take a look at this and figure out what's really going on, what frequency is actually used on the bus in the end.

Technically though, you could just set the speed of the bus before starting display updates or accessing the nRF24L01+.

sedwano, 2020-02-09:
I guess the 2 librarles interfere. I did nlt try the div2 without the display activated. Nor did i try with both on same spi again. I intend to do so soon

sedwano, 2020-02-10:
I did some more experimenting and found more or less the same as Mariusz1, that the compiler has not accepted SPI_HAS_TRANSACTIONS. This is why changing the SPI speed did not work for RF24, and instead it used the clock divider, which it seems was based on the SPI bus of the ILI9488  using 24MHz. 

When I went in to try and set the speed, I noticed that the compiler accepted the #define RF24_SPI_TRANSACTIONS in RF24_config.h but had "forgotten" this in the RF24.cpp.  the behavious was very strange and unpredictable between the *.h files and the rf24.cpp. There was another RF24.h that was part of a different package digikey which i deleted. 

Anyhow. Now both the nRF24L01 and the TFT ILI9488 are working like a charm on the same SPI bus, with what should basically be the original library RF24 code except the forced activation suggested by Mariusz1. 

```C++
#if (defined (SPI_HAS_TRANSACTION) || defined (ESP32)) && !defined (SPI_UART) && !defined (SOFTSPI)
//#if defined (SPI_HAS_TRANSACTION) && !defined (SPI_UART) && !defined (SOFTSPI)
  #define RF24_SPI_TRANSACTIONS
```
I looked around at the ILI9488 *.cpp code and there the define below is "accepted" 
#ifdef SPI_HAS_TRANSACTION

In the exact same line in RF24_config.h is not :-(. Moving it from RF24_config.h to RF24.cpp and it is accepted suddenly. I am using platformIO on VScode. 

My two cent conclusion then is that the evaluation of  SPI_HAS_TRANSACTION for some reason needs to be in the *.cpp file.

Avamander, 2020-02-10:
I'll try pinging a few contributors on the ESP32 core, this really seems like a bug in the core.

Avamander, 2020-02-10:
Hmm, I think I can make the PR myself.

TMRh20, 2020-05-26:
Is this still an issue?
I just tested with ESP32 board `ifdef, ifndef` for `SPI_HAS_TRANSACTION and RF24_SPI_TRANSACTIONS` and it all seems to be working normally.
Board version is v1.0.4
I'm awaiting replacement ESP32 hardware, so can't test fully, but as far as build/compiling, there seems to be a difference.

Mariusz1, 2020-05-27:
> Is this still an issue?

I will try to check and let you know.

Mariusz1, 2020-05-30:
Yes, I still have the problem. Maybe it is related to my PlatformIO environment.

TMRh20, 2020-05-30:
That was with the Arduino IDE so I tried with PlatformIO and got odd results as well.
SPI_HAS_TRANSACTION is defined, seems to produce errors accordingly using the above tests
RF24_SPI_TRANSACTIONS is not getting defined, but if I specifically define it in RF24_config.h without testing for SPI_HAS_TRANSACTION it works normally.  
I then tested with ESP8266. Same results.
Test with Arduino Nano 328. Same results.
The common trend here seems to be PlatformIO.

GabyPCgeeK, 2020-06-30:
Using Arduino IDE 1.8.13 `SPI_HAS_TRANSACTION` is also not detected as defined.
I am using Arduino Uno with AS01-ML01DP5 modules. The modules were not working and I wanted to test using lower SPI speed because the modules have 1K resistors on the I/O. After doing some tests I found that modifying `_SPI.setClockDivider` on RF24.cpp lowered the speed instead of modifying `RF24_SPI_SPEED` in RF24_config.h. After some tests I found out that adding `#include <SPI.h>` before
```
#if defined (SPI_HAS_TRANSACTION) && !defined (SPI_UART) && !defined (SOFTSPI)
    #define RF24_SPI_TRANSACTIONS
#endif
```
made it work but including `#include <SPI.h>` before `#include "RF24.h"` in the ino file or even defining `#define SPI_HAS_TRANSACTION 1` don't make a difference so I added
```
#if defined(SPI_HAS_TRANSACTION) && !defined (SPI_UART) && !defined (SOFTSPI)
    #undef RF24_SPI_TRANSACTIONS
    #define RF24_SPI_TRANSACTIONS
#endif
```
to the end of the
```
#else //Everything else
    #include <Arduino.h>
```
block in which SPI.h is included at least for Arduino Uno and it now works with SPI Transactions.

TMRh20, 2020-07-09:
OK, I think I've got this figured out now, testing OK on platformIO and Arduino IDE with multiple boards.

Should have noticed earlier, but the SPI_HAS_TRANSACTION detection needs to be done at the end of RF24_config.h so that the SPI.h file is included, and SPI_HAS_TRANSACTION is properly defined.

Avamander, 2020-07-09:
> Should have noticed earlier, but the SPI_HAS_TRANSACTION detection needs to be done at the end of RF24_config.h so that the SPI.h file is included, and SPI_HAS_TRANSACTION is properly defined.

Hm. I think all of the examples include SPI.h before RF24 though, there's an intentional build process difference between different architectures that causes the defines to get lost somewhere?

TMRh20, 2020-07-09:
@Avamander I dunno lol??? It does _usually_ work for me if SPI.h is included before RF24.h in the main sketch, however including SPI.h in the main sketch should not be required. RF24.h includes RF24_config.h includes SPI.h, so this should ensure it always works.

TMRh20, 2020-07-16:
Closing as RF24 v1.3.6 should be available via Lib manager and PIO now

Please leave a comment if there are any further related issues.

closed 2020-07-16 by TMRh20

mark1spec, 2021-04-07:
I can confirm, that the problem is fixed now. Thank you :)