nRF24/RF24 #509: ESP32 write always returns 0

toogooda, 2019-07-24:
Hi I have had no problems with Atmega328P but not working with ESP32.

isChipConntected returns 1

Details show as:
```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x3130303030 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x3130303030
RX_PW_P0-6	 = 0x20 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x27
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
```

Code running works on Atmega version with just different pins:

```
#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#define SENDLED 16

RF24 radio(8,5);
const byte address[6] = "00001";

void setup() {
  Serial.begin(115200);
  pinMode(SENDLED,OUTPUT);
  radio.begin();
  radio.isChipConnected();
  radio.openWritingPipe(address);
  radio.setPALevel(RF24_PA_MAX);
  radio.setDataRate(RF24_250KBPS);
  radio.printDetails();
  radio.stopListening();
}

void loop() {
  digitalWrite(SENDLED,HIGH);
  const char text[] = "nrftest";
  Serial.println(text);
  byte a = radio.write(&text, sizeof(text));
  Serial.print("Send Result:");
  Serial.println(a);
  digitalWrite(SENDLED,LOW);
  delay(2000);
}

```
Tracked it down to while loop in write function
`while( ! ( get_status()  & ( _BV(TX_DS) | _BV(MAX_RT) ))) {`

Outputs that show how the above evaluated:
get_status():14
_BV(TX_DS):32
_BV(MAX_RT):16
( _BV(TX_DS) | _BV(MAX_RT) ):48
 get_status()  & ( _BV(TX_DS) | _BV(MAX_RT) ):0

It always = true so always returns 0 
```

#if defined (FAILURE_HANDLING)
          printDetails();
				  return 0;		
```
Does this sound like software or hardware?

toogooda, 2019-07-24:
I have 4 decoupling caps .1uf 1uf 10uf 100uf and have checked every wire with multimeter

TMRh20, 2019-07-25:
Hard to say. I would suggest testing with another ESP32 if possible, double check for proper connections etc.

toogooda, 2019-07-26:
> Hard to say. I would suggest testing with another ESP32 if possible, double check for proper connections etc.
> […](#)

OK will try another ESP I suppose the solder on the esp pins could also be an issue.
Failing that I can solder another cap directly to the nRF module.
I will chuck it on the scope and see what vcc is doing

Can you please explain what `while( ! ( get_status() & ( _BV(TX_DS) | _BV(MAX_RT) ))) {`
is doing?

and what values of 
`get_status():14
_BV(TX_DS):32
_BV(MAX_RT):16`

means?

wmarkow, 2019-07-26:
I see that the printed details look correct. Also **isChipConnected** return **true**, so there is no hardware issue here.

>Can you please explain what while( ! ( get_status() & ( _BV(TX_DS) | _BV(MAX_RT) ))) {
is doing?

It waits until one of the two conditions is met:
 * either the data are sent to the air and ACK is received (if ACK is anabled - but in your code it is enabled by default by RF24 library). **TX_DS** bit of STATUS register tells us about this.
 * or the max retransmition has been reached and unfortunatelly the data has not been probably not sent to the recipient and there was no ACK from the recipient. About this tells us **MAX_RT** bit of the STATUS register. By default in RF24 library the number of retransmitions is set to **15** and a timeout for geting ACK back from recipient is 1500us.

If you have **FAILURE_HANDLING** defined, then this **while** loop can be broken after 95ms. In your case it means that the chip didn't finish even those 15 retransmitions tries. We assume that you do not have any hardware issues. So the whole sending process (togehter with ACK handling and retransmitions) takes more than 95ms. Is that even possible? With 250kbps of transmition speed, 15 retries, packet length of around 40B.

Some general question:
 * do you try to send data between Arduino Uno and ESP32?
 * or you just send the data "in the air" without any receiver. Then the **write** should fail in both boards.
 * I'm not sure if you correctly configured the writing/reading pipes/channels. You need correctly configure them for the transmitter and receiver.

toogooda, 2019-07-26:
> Is that even possible? With 250kbps of transmition speed, 15 retries, packet length of around 40B.
would work out at < 3ms plus 15 times whatever the wait time is between retries...

> do you try to send data between Arduino Uno and ESP32?
Reciever is a Nano (Atmega328P) Sender is a ESP32. The sketches send and receive are from a nano to nano implementation that is working, I just changed the pins for the ESP. When I turn on the old nano sender it it proves the receiver is still working.

> I'm not sure if you correctly configured the writing/reading pipes/channels. You need correctly 
configure them for the transmitter and receiver.

Here is the receiver code which works for when nano is sending

`#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

#define LED 21

RF24 radio(8, 5);
//const int buzzer = 7;
const byte address[6] = "00001";

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println("Setup");
  //pinMode(buzzer, OUTPUT);
  pinMode(LED,OUTPUT);
  radio.begin();
  if (radio.isChipConnected()){
    Serial.println("nrf24 Chip Detected");
  }else{
    Serial.println("NO! nrf24 Chip Detected");
  }
  radio.openReadingPipe(0, address);
  radio.setPALevel(RF24_PA_MAX);
  radio.setDataRate(RF24_250KBPS);
  radio.startListening();
  Serial.println("Listening");
}

void loop() {
  char text[11] = "";

  if (radio.available()) {
    Serial.println("Recieving");
    radio.read(&text, sizeof(text));
    String transData = String(text);
    Serial.println(transData);
 //   if (transData == "nrftest") {
      Serial.println(transData);
      //tone(buzzer, 1000);
      digitalWrite(LED, HIGH);
      delay(200);
      //noTone(buzzer);
      digitalWrite(LED, LOW);
//    }
  }
`

toogooda, 2019-07-26:
OK interesting thing I have noticed when i hit reset on ESP32 it successfully sends 1-3 packets like they were queued up waiting to go and didn't have power untill the ESP CPU was shut down.
Is there hardware queueing on the nRF chips?
So sounds like a power issue to me I need to check my schematic to see if the caps near by actually have a rout to the nRF24 and not going via regulator or something like that.

Also I wonder if the USB/ESP regulator is enough to power both esp32 and nRF24 in max mode.

wmarkow, 2019-07-26:
The chip has a FIFO transmition queue that is able to take up to three packets.

In the [datasheet](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf) at the end of section 7.5.1 is written something like:
>If the TX FIFO (PRX) contains more than one payload to a PTX, payloads are handled using the first in – first out principle. The TX FIFO (PRX) is blocked if all pending payloads are addressed to a PTX where the link is lost. In this case, the MCU can flush the TX FIFO (PRX) by using the FLUSH_TX command.

I understand this in the following way: if you have auto ACK enabled (which is enabled by default in RF24 library) and if your transmitter finds out that the link to the receiver has been broken then the TX FIFO queue will not be flushed automatically and it will be blocked, so it will not send actually any new packets. You need to call **flush tx** manually in your code.

The 3.3V power supply is crutial. I think I had some issues in the past with on board 3.3 voltage regulater on Arduino Nano. The 3.3V was a part of FTD232 chip and it was able to deliver like 50mA which was not enough to transmitt packets by the nRF24L01 chip. I needed to use some external 3.3V power source. Please check it in your board - that may be an issue.

Avamander, 2019-07-26:
> The 3.3V power supply is crutial. I think I had some issues in the past with on board 3.3 voltage regulater on Arduino Nano. The 3.3V was a part of FTD232 chip (or something similar) and it was able to deliver like 50mA which was not enough to transmitt packets by the nRF24L01 chip.

Yeah it can not be understated how tedious it can be to debug issues related to power issues, in this case though the decoupling should be roughly enough. Though when I've designed boards myself I've decoupled both 5V and 3.3V with 5mF electrolytic on both, plus 10uF ceramic capacitor just in case.

wmarkow, 2019-07-26:
Just one more question:
 1. what is the ```sizeof(char)``` on Nano
 2. what is the ```sizeof(char)``` on ESP32

Is it one byte in both cases or it it 2 bytes on ESP32? You send send a ```char``` data. Yoy may have some "integration" issue between both boards.

toogooda, 2019-07-26:
> Just one more question:
> 
> 1. what is the `sizeof(char)` on Nano
> 2. what is the `sizeof(char)` on ESP32
> 
> Is it one byte in both cases or it it 2 bytes on ESP32? You send send a `char` data. Yoy may have some "integration" issue between both boards.
> 
> EDIT:
> Arduino Nano **char** is 1B length: https://www.arduino.cc/reference/en/language/variables/data-types/char/
> For ESP32 it seems to be also 1B length: [espressif/arduino-esp32#1745](https://github.com/espressif/arduino-esp32/issues/1745)

Thanks yes I remember when I did a Nano -> ESP8266 got caught out sending a struct where data types meant something different.

toogooda, 2019-07-26:
> The chip has a FIFO transmition queue that is able to take up to three packets.
> 
> In the [datasheet](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf) at the end of section 7.5.1 is written something like:
> 
> > If the TX FIFO (PRX) contains more than one payload to a PTX, payloads are handled using the first in – first out principle. The TX FIFO (PRX) is blocked if all pending payloads are addressed to a PTX where the link is lost. In this case, the MCU can flush the TX FIFO (PRX) by using the FLUSH_TX command.
> 
> I understand this in the following way: if you have auto ACK enabled (which is enabled by default in RF24 library) and if your transmitter finds out that the link to the receiver has been broken then the TX FIFO queue will not be flushed automatically and it will be blocked, so it will not send actually any new packets. You need to call **flush tx** manually in your code.
> 
> The 3.3V power supply is crutial. I think I had some issues in the past with on board 3.3 voltage regulater on Arduino Nano. The 3.3V was a part of FTD232 chip (or something similar) and it was able to deliver like 50mA which was not enough to transmitt packets by the nRF24L01 chip. I needed to use some external 3.3V power source. Please check it in your board - that may be an issue.

OK that makes sense then that they were still sitting in a queue. It seems to be all power related this time.

toogooda, 2019-07-26:
OK @Avamander after your comment I had another think about my decoupling and I think I have the issue let me explain here:

I have done a few PCB designs but still learning, In the past I always did a GND plane on the bottom and VCC plane on the top so whenever you placed a cap near an item it was decoupled well.
I recently read articles that told me this was not best practice and I should do both layers as GND with lots of vias to ensure GND is shortest path for everything.
So my new PCB has 2 GND planes and as I was lazy and used auto router this time the caps although near the nRF24 module actually are not connected apart from all the way back at the LDO meaning the nRF24 has no decoupling at all :(

I will try prove this by soldering directly from nRF24 Vcc to the bank of caps right next to it and if it solves it then its confirmed as a hardware issue.

wmarkow, 2019-07-26:
In my solution I have soldered one electrolytic capacitor directly on the RF24 chip, right between GND and VCC pins. It works.

Avamander, 2019-07-27:
> I recently read articles that told me this was not best practice and I should do both layers as GND with lots of vias to ensure GND is shortest path for everything.

The way you've done it is being done it previously a lot, in many cases it doesn't really matter.

> As a result the caps although near the nRF24 module actually are not connected apart from all the way back at the LDO meaning the nRF24 has no decoupling at all :(

Depending on the trace length and environmental noise this might not be an issue, you can keep large electrolytic caps at the regulator but add one small ceramic next to the radio itself and it should filter out any HF noise the traces might pick up.

toogooda, 2019-07-28:
> In my solution I have soldered one electrolytic capacitor directly on the RF24 chip, right between GND and VCC pins. It works.

Waked a massive 1000uf cap directly to Vcc and Gnd but didn't fix it however if I leave it running every 5-10 mins one is received.  So in total i have 1000uf direct to pins plus (1000, 100, 10, 1 and 0.1 uf caps less directly) Surely any more is excessive?

toogooda, 2019-07-28:
> Depending on the trace length and environmental noise this might not be an issue, you can keep large electrolytic caps at the regulator but add one small ceramic next to the radio itself and it should filter out any HF noise the traces might pick up.

Starting to run out of ideas, the Atmega328P PCB works 100% with the same module and same code. The only other PCB difference I can think of is that I had no copper plates at all below the module with it being a SMD version I thought it might cause interference. but I have two GND layers under the non working version.  This is the specific nRF24 I am using its better quality FCC approved etc [nRF24SMD](https://www.aliexpress.com/item/32993806214.html?spm=a2g0s.9042311.0.0.27424c4db7W6Gr)
Given it has an external aerial I didn't bother this time.

toogooda, 2019-07-29:
OK I have tried everything hardware power related many caps lipo not usb powered removed components and proved the work on Nano and stm32 versions. The only time this one sends is when I hit rst and it dumps the queue for a few milliseconds.
Has anyone actually ever tested ESP32 with this library and did it every work if so what version and did they use VSPI or HSPI?
I am out of ideas Why does it work for ATmega/SMT/8266 but fail with ESP32

toogooda, 2019-07-30:
SOLVED IT
In the end I bread boarded it to take all the possible PCB issues out of it. And got the same result. I thought about the only steps people could do different wh got it working and realised CE and CSN were randomly seleted by me as spare GPIO's toI changed CE from GPIO8 to GPIO4 and it worked right away.
After looking through the datasheet I can't find a reason why GPIO8 world not work it does have a dual function as SDIO but can't see why that wouldn't make a difference. Anyway the moral of the story is breadboard your solution before PCB design .
That you all for your help with this. I have learned a lot more about the library as a result.
Cheers Andrew

closed 2019-07-30 by toogooda

toogooda, 2019-07-30:
For anyone else with this problem see this article [Here](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/)

GPIO | Input | Output | Notes
-- | -- | -- | --
0 | pulled up | OK | outputs PWM signal at boot
1 | TX pin | OK | debug output at boot
2 | OK | OK | connected to on-board LED
3 | OK | RX pin | HIGH at boot
4 | OK | OK |  
5 | OK | OK | outputs PWM signal at boot
6 | x | x | connected to the integrated SPI flash
7 | x | x | connected to the integrated SPI flash
8 | x | x | connected to the integrated SPI flash
9 | x | x | connected to the integrated SPI flash
10 | x | x | connected to the integrated SPI flash
11 | x | x | connected to the integrated SPI flash
12 | OK | OK | boot fail if pulled high
13 | OK | OK |  
14 | OK | OK | outputs PWM signal at boot
15 | OK | OK | outputs PWM signal at boot
16 | OK | OK |  
17 | OK | OK |  
18 | OK | OK |  
19 | OK | OK |  
21 | OK | OK |  
22 | OK | OK |  
23 | OK | OK |  
25 | OK | OK |  
26 | OK | OK |  
27 | OK | OK |  
32 | OK | OK |  
33 | OK | OK |  
34 | OK |   | input only
35 | OK |   | input only
36 | OK |   | input only
39 | OK |   | input only