nRF24/RF24 #511: NRF Wont connect with arduino or esp32.

hazee007, 2019-07-30:
Hello guys i have been having this same problem for days now
i am certain that my connections are correct cause i have connected with two different arduino uno with nrf24 as well as arduino nano and also esp32 and all are not working.. i noticed that if i touch a the MISO wire i gives so output like this
Sent response 1056964608
Sent response 0
Sent response 16777216
Sent response 1061101312
Sent response 0
Sent response 0
Sent response 0
Sent response 514
Sent response 0
Sent response 1061101375
Sent response 0
Sent response 0
Sent response 17776399
Sent response 0
Sent response 33554432
Sent response 1057030671
Sent response 0
Sent response 1061109511
Sent response 16777216
Sent response 2134843199
Sent response 117653251
Sent response 0
Sent response 1065320255
Sent response 16777216
Sent response 2139094847
Sent response 251658240

and for transmitting part

Failed, response timed out.
Now sending
Failed, response timed out.
Now sending
Failed, response timed out.
Now sending
Failed, response timed out.
Now sending
Failed, response timed out.
Now sending
Failed, response timed out.
Now sending
Failed, response timed out.
Now sending

this is my printDetails() output

RF24/examples/GettingStarted/

ROLE: Pong back

*** PRESS 'T' to begin transmitting to the other node

STATUS	= 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	� = 0x0000000000 0x0000000000
RX_ADDR_P2-5	� = 0x00 0x00 0x00 0x00
TX_ADDR	= 0x0000000000
RX_PW_P0-6	� = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA	= 0x00
EN_RXADDR	� = 0x00
RF_CH	= 0x00
RF_SETUP	� = 0x00
CONFIG	= 0x00
DYNPD/FEATURE	� = 0x00 0x00
Data Rate	= 1MBPS
Model	= nRF24L01
CRC Length	= Disabled
PA Power	= PA_MIN
Now sending 0 as payload

Got response 0, round-trip delay: 197645340 microseconds

Now sending 173 as payload

This means that the wiring is correct, ill be glad if anyone can help me with this.

Sempailik, 2019-11-21:
I have same problem. 
It is interesting that if I put the same code in Arduino Nano it works and when it is in ESP8266 not. I used the same v1.3.3 library. I have read that, for example, RPi responds before Arduino on the receiving side can switch to listening. They write at least 130us from switching to the next message sending.
So I thought that ESP would run faster at 80MHz, but it wasn't that true.
 When I measured the time between receiving and sending a message (like ACK), switching Arduino nano took about 2200us and ESP8266 it took 3800us. I also tried the older ESP v1.1.7u library and downloaded this time to 1900us (using only a different version of the library). Still, the other device did not receive the message.
Oh, and the automatic ACK is forbidden.

Sempailik, 2019-11-21:
By listing what you put in there (radio.printDetails ();) it looks like you have the CE and CSN pins wrong. I had them swapped these two pins and it wrote to as yours.

Avamander, 2020-01-28:
Yeah, this is ordinarily caused by an issue in wiring. If you still have this issue and checking your wiring didn't help, just reply.

closed 2020-01-28 by Avamander