nRF24/RF24 #512: Raspberry Pi 3 - Receiving not working

kowlown, 2019-08-02:
I have a Java Wrapper calling each second a c++ method to listen a message.
Alas, I don't manage to receive any message sent from my Arduino Nano.

Raspberry code : 
`
	#include "receiver.h"

	RF24 radio(RPI_V2_GPIO_P1_22, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);

	const uint8_t data_pipe[6] = "00001";
	bool working = false;

	using namespace std;

	//Initialization
	void setup() {

		cout << "setup begin" << endl;
		radio.begin();
		cout << "set retries" << endl;
		radio.setRetries(15, 15);
		cout << "set PA Level" << endl;
		radio.setPALevel(RF24_PA_MAX);
		cout << "open reading pipe" << endl;
		radio.openReadingPipe(1, data_pipe);
		radio.printDetails();

	}

	//We want to listen and wait from a message if have one or return an empty string.
	//It's synchronous.
	char *listen() {
		cout << "start listening" << endl;
		radio.startListening();

		cout << "listen" << endl;
		chrono::steady_clock::time_point start = chrono::steady_clock::now();
		bool timeout = false;

		// While nothing is received
		while (!radio.available()) {
			chrono::steady_clock::time_point end = chrono::steady_clock::now();
			auto duration = chrono::duration_cast<chrono::seconds>(end - start);
			usleep(5000); //Sleep for 5 ms
			// If waited longer than 1 seconds, indicate timeout and exit while loop
			if (duration.count() > 1) {
				timeout = true;
				break;
			}
		}

		cout << "Time timeout:" << timeout << endl;

		if (timeout) {                                             // Describe the results
			cout << "Failed, response timed out." << endl;
		} else {
			cout << "getDynamicPayloadSize" << endl;
			int payload_size = 0;
			for (int i = 0; (payload_size <= 0 && i < 1000); i++) {
				payload_size = radio.getDynamicPayloadSize();
				usleep(5000);  //Sleep for 5 ms
			}

			if (payload_size > 1) {
				cout << "char *payload = new char[payload_size + 1];" << endl;
				char *payload = new char[payload_size + 1];
				radio.read(payload, payload_size);
				cout << "Read" << endl;
				payload[payload_size] = '\0';
				cout << "stop listening" << endl;
				radio.stopListening();
				return payload;
			}
		}
		cout << "stop listening" << endl;
		radio.stopListening();
		return '\0';
	}
`

I have this message : 

> JNI setup
> setup begin
> set retries
> set PA Level
> open reading pipe
> ================ SPI Configuration ================
> CSN Pin          = CE0 (PI Hardware Driven)
> CE Pin           = Custom GPIO25
> Clock Speed      = 8 Mhz
> ================ NRF Configuration ================
> STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
> RX_ADDR_P0-1     = 0xa8a8e1fc00 0x3130303030
> RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
> TX_ADDR          = 0xffffffffff
> RX_PW_P0-6       = 0x00 0x20 0x00 0x00 0x00 0x00
> EN_AA            = 0x01
> EN_RXADDR        = 0x03
> RF_CH            = 0x4c
> RF_SETUP         = 0x07
> CONFIG           = 0x0e
> DYNPD/FEATURE    = 0x00 0x00
> Data Rate        = 1MBPS
> Model            = nRF24L01+
> CRC Length       = 16 bits
> PA Power         = PA_MAX
> Start read
> JNI listen
> start listening
> listen
> Time timeout:1
> Failed, response timed out.
> stop listening
> End read=> message:null
> null
> Start read
> JNI listen
> start listening
> listen
> Time timeout:1
> Failed, response timed out.
> stop listening
> End read=> message:null
> null
> Start read
> JNI listen
> start listening
> listen
> Time timeout:0
> getDynamicPayloadSize
> payload size : 14
> Read
> stop listening
> payload:
> End read=> message:
> 
> Start read
> JNI listen
> start listening
> listen
> Time timeout:0
> getDynamicPayloadSize
> stop listening
> End read=> message:null
> null
> Start read
> JNI listen
> start listening
> listen
> Time timeout:0
> getDynamicPayloadSize
> stop listening
> End read=> message:null
> null
> Start read
> JNI listen
> start listening
> listen
> Time timeout:1
> Failed, response timed out.
> stop listening
> End read=> message:null
> null
> Start read
> JNI listen
> start listening
> listen
> Time timeout:1
> Failed, response timed out.
> stop listening
> End read=> message:null
> null

closed 2019-08-13 by kowlown

MaadMind, 2019-10-30:
Hi, did you fix this issue?

kowlown, 2019-11-04:
I decided to not use dynamic payload but instead I used a struct with float for the temperature, humidity and pressure.