nRF24/RF24 #515: nrf24l01 works with arduino uno and nano, but not pro mini

NghiDao2, 2019-08-14:
I used a 3.3v nrf24l01 for a arduino and a nano and it worked perfectly fine, however with my arduino promini it doesnt work, but instead spams data recieved. Even if i unplug everything, the arduino promini still thinks its recieving data, but this doesn't happen to the arduino uno and nano.

code i used: https://forum.arduino.cc/index.php?topic=421081

output (with the nrf24l01 plugged in and unplugged is the same):

Data received
Data received
Data received
Data received
Data received
...

I believe this may be something to do with the arduino pro mini rather than the nrf24l01 because it worked fine with the arduino uno, and it spams "Data received" even with the nrf2l01 unplugged.

Please help
Thanks!

soligen2010, 2019-08-14:
Try other pro-minis purchased from a different suppliers.  I have made many units with pro-minis and have had a couple bad ones that seemed fine for other things, but just would not work with NRF24L01.  It was just 2 units that I think i got from the same source (direct from china).

Another thing you can try is adding your own 10k pullup resistors to the lines (or turn on pullups in the code if it isn't).

NghiDao2, 2019-08-14:
ok i will order a new promini an ill reply once it arrives

Avamander, 2019-08-14:
Decouple the power bus with large electrolytic capacitors, the regulators on cheap Pro Minis are really bad.

NghiDao2, 2019-08-16:
ive bought a new arduino promini, and i still get the same result
it constantly output "data recerved" even when nothing is connected to it

Avamander, 2019-08-16:
Use the example code and see what that does.

NghiDao2, 2019-08-16:
ive tried a few codes but i still get the same result, it constantly says data is received

TMRh20, 2020-08-11:
Not much we can do about the behaviour when the device is disconnected. The lib has been tested and confirmed to work with pro minis. This has to be a wiring &/or hardware issue.

closed 2020-08-11 by TMRh20