nRF24/RF24 #517: running on an RPI4 produces "bcm2835_init: gpio mmap failed: Cannot allocate memory" error

thncode, 2019-08-17:
after searching and testing all day I have to give up for now. moving a working environment from a RPI3 to a RPI4 does not work but produces the error "bcm2835_init: gpio mmap failed: Cannot allocate memory" even after radio.begin() - this seems to be a memory problem either with the new RPI4 or with its 4GB memory...

Any ideas?

TMRh20, 2019-08-17:
Have you tried using the SPIDEV driver? (Run configure with —driver=SPIDEV before building)

TMRh20, 2019-08-17:
See https://groups.google.com/forum/#!topic/bcm2835/BwZXVsDRtwI

BCM driver just updated per thread

thncode, 2019-08-18:
yes - this worked! Thx a lot!

closed 2019-08-18 by thncode

Paul3366-Goodman, 2020-01-15:
I got the same same error when I test my new rpi4 with pan-tilt hat.

May you give some hints how to solve it?

TMRh20, 2020-01-16:
Have you tried the SPIDEV driver per the previous comments?

Paul3366-Goodman, 2020-01-17:
Hi TMRh20,  pardon me as a newbie.  I don't know how to run configure with spidev driver.  Please kindly point to the right to run config with spidev driver.

Avamander, 2020-01-17:
@Paul3366-Goodman 
> Hi TMRh20, pardon me as a newbie. I don't know how to run configure with spidev driver. Please kindly point to the right to run config with spidev driver.

https://github.com/nRF24/RF24/issues/517#issuecomment-522272992

srinivasank, 2020-03-21:
pi@raspberrypi:~/bcm2835-1.58 $ sudo ./configure --driver=SPIDEV
configure: error: unrecognized option: `--driver=SPIDEV'
Try `./configure --help' for more information

TMRh20, 2020-03-21:
@srinivasan Your error message indicates you are running configure from ~/bcm2835-1.58 directory... if you want to configure and compile rf24 you should switch to the rf24 directory...