nRF24/RF24 #518: Issue with RPi 3 B+, stuck at radion.begin() at program relaunch

kowlown, 2019-08-21:
Hello,
I manage to run and execute the cpp examples but when I exit and relaunch the application,
I'm stuck at radio.begin(); without any error message.
I have to reboot my raspberry to be able to make the program receive message again.

Avamander, 2019-08-23:
You need to make your program exit nicely, release SPI devices and close files otherwise you will have issues.

kowlown, 2019-08-23:
@Avamander  To release SPI I have to call to RF24.stopListening and RF24.closeReadingPipe?

TMRh20, 2019-08-24:
Have you tried using the SPIDEV driver? Run ./configure —driver=SPIDEV  then make and sudo make install from the RF24 directory. I’d be curious to see if it happens with a different driver.

closed 2020-03-29 by TMRh20