nRF24/RF24 #523: how to Change TX_ADDR?

imohammad74, 2019-09-01:
I coding a Arduino's program. it worked well but after some day it break and don't send or receive data. I printed the NRF24L01 configuration and result was:
```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f0d2 0xf0f0f0f0e1
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f0d2
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
```
I
I run a sample code "pingpair_dyn.ino" in get NRF configuration details.So compare between ttwo result
"pingpair_dyn.ino"
```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f0e1 0xf0f0f0f0d2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f0e1
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
```
The TX_ADDR  is different in this results. 
how to change TX_ADDR in my program ? 
Thanks

TMRh20, 2019-09-03:
Use openWritingPipe per http://tmrh20.github.io/RF24/classRF24.html#af2e409e62d49a23e372a70b904ae30e1 to change tx addr.

imohammad74, 2019-09-04:
What's it "CONFIG"?
I changed the pipes and TX_ADDR changed  and corrected .
But now CONFIG is difference between them.
I have generally question:
Should "NRF Configuration" be  same at Raspberry (3b+) and Arduino Uno?

TMRh20, 2019-09-06:
> What's it "CONFIG"?

Its the hex value of the Config register of the nrf chip. See page 54 of [here](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf)

> Should "NRF Configuration" be same at Raspberry (3b+) and Arduino Uno?

Not necessarily, especially if one is in RX mode and one is in TX mode.

closed 2020-03-29 by TMRh20