nRF24/RF24 #524: pyRF24- Makefile.inc is missing (?)

architeuthisss, 2019-09-06:
While following the directions on [http://tmrh20.github.io/RF24/Python.html](http://tmrh20.github.io/RF24/Python.html) for building/installing the Python wrapper library, I was unable to build the library. On further inspection, I found that in the Makefile there is referenced a Makefile.inc. 
To the best of my knowledge, Makefile.inc does not exist in this repository and is not generated before building the library. 

In Makefile, Makefile.inc is referenced here:
`CONFIG_FILE=Makefile.inc
REMOTE_ERROR="[ERROR] Remote machine not configured. Run configure with respective arguments."

include $(CONFIG_FILE)`

[image]

TMRh20, 2019-09-06:
Did you build/install the library normally first? I think that’s where the makefile.inc gets generated.

architeuthisss, 2019-09-07:
I tried that, still having issues, although they don't appear to be directly related to Makefile.inc anymore. I'll keep looking into it.

architeuthisss, 2019-09-13:
Looks like the problem was I had not installed the C library prior to installing the Python library.

closed 2019-09-13 by architeuthisss