nRF24/RF24 #525: ARM64 (JETSON TX2) not working...

chipscal, 2019-09-09:
Dear all,

I'm trying to make the library work with a JETSON TX2 board but I have some issues.
I've modified the configure script and added a soc with the following flags:
  -->> flags="-march=armv8-a  -mtune=cortex-a57" (-mfpu=neon-vfpv4 -mfloat-abi=hard are not recognized by the compiler and SIMD is defaut in ARM64 i think)
I'm using the SPIDEV driver.

I'm able to correctly compile and link the library but the module wont configure correctly and obviously do not send/receive data.

Please note:
     -> I've already tested the modules with two RPi3, using SPIDEV, and they work correctly.
     -> the spidev driver itself is working correctly (I've jumped the MISO and MOSI pin and made a test)
    ->  the gpios seems also to work well (able to rise up the pin that I'm using via the device file)
    -> I've used as cepin=388 (gpio388) and cspin=30 (/dev/spidev3.0)

I'm starting to think that the malfunctioning may be related to some issue with the 64 bit architecture of the board.. Have someone ever tried to build on arm64 machine?? 
Do you have some suggestion?? Am I missing something??

Thanks for the help,
Luca

Avamander, 2019-09-10:
It's possible that there's some kind of a data type size mismatch. However if you connect the module to your board and write a sample script that just reads the module's configuration (you can see which registers contain what from either RF24's source or from the datasheet), does that work? Also how does the library not work for you? If it hangs, please determine where exactly.

Also just mentioning that if it's GCC you can just do `-march=native -mtune=native` and you shouldn't need to provide any other flags.

chipscal, 2019-09-10:
If I try to configure the radio on a RPi3 I obtain this: (correct)

STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5     = 0x33 0xce 0x3e 0xe3
TX_ADDR          = 0x65646f4e32
RX_PW_P0-6       = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA            = 0x3e
EN_RXADDR        = 0x3f
RF_CH            = 0x5a
RF_SETUP         = 0x0d
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x3f 0x04
Data Rate        = 2MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_HIGH

Same code in the Jetson I obtain a complete different configuration: (wrong)

STATUS           = 0x03 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=1 TX_FULL=1
RX_ADDR_P0-1     = 0x222222e2e0 0x222222220e  
RX_ADDR_P2-5     = 0x88 0x23 0x8f 0x70
TX_ADDR          = 0x10b0170388
RX_PW_P0-6       = 0x00 0x80 0x00 0x80 0x00 0x80
EN_AA            = 0x07
EN_RXADDR        = 0x80
RF_CH            = 0x05
RF_SETUP         = 0xef
CONFIG           = 0x80
DYNPD/FEATURE    = 0x8f 0x00
Data Rate        = 1MBPS  <<<---- It should be 2
Model            = nRF24L01    <<<--------- Note: It detects a wrong module! 
CRC Length       = 8 bits  <<<-------- I set it to 16
PA Power         = PA_MAX  <<<--- I set it to HIGH

If I try to receive from jetson it simply doesn't receive nothing... If I try to send it shows me the fail message where it suggest that the wiring may be wrong...

Today I will try to set some registers and read it back using the provided functions and I will let you know!

Thanks for the help!

Avamander, 2019-09-14:
Can you use barebones SPI and see if that still has the issue? Knowing if the problem still exists or doesn't would narrow it down to read_register.

Avamander, 2019-09-14:
I just personally suspect it's some lower-level issue because the one-bit-shift issue hasn't occurred before.

Avamander, 2019-09-14:
I suspect there's some kind of SPI bus misconfiguration, you can read about the different bus configuration possibilities here https://arduino.stackexchange.com/a/13092/11529 but I don't have a Jetson to provide you sample code to set a specific mode.

Avamander, 2019-09-24:
The mode Arduinos use for SPI is MODE0:

Mode | Clock Polarity (CPOL) | Clock Phase (CPHA) | Output Edge | Data Capture
-- | -- | -- | -- | --
SPI_MODE0 | 0 | 0 | Falling | Rising

You should try setting CPHA and CPOL to 0, I wouldn't change the speed because it's not really the issue here.

tantinlala, 2020-08-30:
I believe I might have found a relevant bug while working with the Jetson Nano. I was also running into an issue where my pipe addresses were not being set correctly.

In RF24/utility/SPIDEV/spi.cpp, it seems that in SPI::transfernb, cs_change is set =0 by default, which would mean that the chip would not be automatically "hardware deselected" after every transfer. however, after I set the default value of cs_change to 1, I was able to have the pipe addresses set correctly and run the pingpair_dyn.py script successfully.

TMRh20, 2020-08-31:
Not sure about this one. 

https://docs.huihoo.com/doxygen/linux/kernel/3.7/structspi__transfer.html

> All SPI transfers start with the relevant chipselect active. Normally it stays selected until after the last transfer in a message. Drivers can affect the chipselect signal using cs_change.
(i) If the transfer isn't the last one in the message, this flag is used to make the chipselect briefly go inactive in the middle of the message. Toggling chipselect in this way may be needed to terminate a chip command, letting a single spi_message perform all of group of chip transactions together.
(ii) When the transfer is the last one in the message, the chip may stay selected until the next transfer. On multi-device SPI busses with nothing blocking messages going to other devices, this is just a performance hint; starting a message to another device deselects this one. But in other cases, this can be used to ensure correctness. Some devices need protocol transactions to be built from a series of spi_message submissions, where the content of one message is determined by the results of previous messages and where the whole transaction ends when the chipselect goes intactive.
The code that submits an spi_message (and its spi_transfers) to the lower layers is responsible for managing its memory. Zero-initialize every field you don't set up explicitly, to insulate against future API updates. After you submit a message and its transfers, ignore them until its completion callback.
Definition at line 492 of file spi.h.

You may be able to add a #define to SPI.cpp to detect the Jetson and configure cs_change to 1

TMRh20, 2020-08-31:
Just to be clear, I should node that cs_change 1 does not work on RPi. I don't have a scope to see exactly what is going on.

2bndy5, 2023-04-21:
See https://github.com/nRF24/RF24/issues/876#issuecomment-1518269315

We may be able to close this issue in couple weeks (when Ububtu18.04 hits EoL).

---
> after I set the default value of cs_change to 1, I was able to have the pipe addresses set correctly

This may be specific to the Jetpack SDK since it includes the Linux kernel (which may be customized though not explicitly mentioned in the docs).

---
Reminder to passers-by, the newer [CMake install instructions](https://nrf24.github.io/RF24/md_docs_using_cmake.html) are required for a 64-bit OS

2bndy5, 2025-07-02:
Closing this.

1. The hardware in question is outdated. I don't think nVidia is actively supporting the Jetson TX2 anymore (beyond maintenance fixes in the Jetpack SDK v4.6.x).
2. Using the CMake install instructions should work much better than our old hand-crafted configure script.
3. Its been 6 years since this was opened, and there is no funding or plan to acquire an nVidia TX2 just for exploring this 1 issue. We need someone that still has a TX2 on hand to verify if the latest RF24 lib changes work or not.

closed 2025-07-02 by 2bndy5