nRF24/RF24 #527: RF24 + attiny2313

linuseing, 2019-09-24:
Hi, I tried now for two getting the nrf24 to work with my attiny2313, I have tested my code and modules successfully with an ESP8266.

here the code for the attiny:

`#include <Arduino.h>
#include <nRF24L01.h>
#include <RF24.h>

#define CE_PIN 9
#define CSN_PIN 10
#define __AVR_ATtiny2313__
#define DO = 14
#define DI = 15
#define USCK = 16

const byte thisSlaveAddress[5] = {'R', 'x', 'A', 'A', 'A'};

RF24 radio(0, 1);

char dataReceived[10]; 
bool newData = false;

byte LEDpin = 2;

void setup()
{
  pinMode(LEDpin, OUTPUT);
  for (int i = 0; i < 2; i++)
  {
    analogWrite(LEDpin, 255);
    delay(100);
    analogWrite(LEDpin, 0);
    delay(100);
  }
  delay(1000);
  radio.begin();
  radio.setDataRate(RF24_250KBPS);
  radio.openReadingPipe(1, thisSlaveAddress);
  radio.startListening();
  if (radio.isChipConnected())
  {
    for (int i = 0; i < 2; i++)
    {
      analogWrite(LEDpin, 255);
      delay(100);
      analogWrite(LEDpin, 0);
      delay(100);
    }
  }
}

void getData()
{
  if (radio.available())
  {
    radio.read(&dataReceived, sizeof(dataReceived));
    newData = true;
  }
}

void showData()
{
  if (newData == true)
  {
    if (dataReceived[8] == '0')
    {
      analogWrite(LEDpin, 255);
    }
    else
    {
      analogWrite(LEDpin, 0);
    }

    newData = false;
  }
}

void loop()
{
  getData();
  showData();
}`

it should blink a LED depending on the send message, again it works with my ESP's.
Also in the setup function, I added a blink to see if the module is connected, which it is not.
Does anyone know how to make it work?
Thanks in advance

linuseing, 2019-09-24:
I'm sorry for the ugly code insertion, I used GitHub's "insert code"...

Avamander, 2019-09-24:
I don't have any ATTiny-s so I can't test if it's your code or not, but have you tried using the sending example and see if that works? If not, I'd just try rewiring the thing, it might have a subtle mistake.

linuseing, 2019-09-25:
I tried that, no luck, I also tried it with an attiny84, again no luck...

2bndy5, 2020-12-14:
Seems to me like your missing `#include <SPI.h>`, but I could be mistaken. @linuseing The latest release of  v1.3.10 now depends on the SpenceKonde ATTinyCore which supports the ATTiny2313  & has its own USI-SPI implementation (should still need to `#include <SPI.h>` though). Keep in mind that the ATTiny2313 has the least amount of memory available in the ATTiny family.

ceyhunozgun, 2021-01-19:
Hi,
If I am not mistaken, it seems that Attiny2313 is supported on the latest releases and it is said that the support is given with SpenceKonde ATTinyCore.
I can't compile rf24ping85 example for Attiny2313a with SpenceKonde ATTinyCore, it says that 'text section exceeds available space in board'. The compiled program is 2816 bytes which exceeds available 2048 bytes with LTO enabled.
I am using Arduino 1.8.13 on Windows and RF24 1.3.11.

Am I missing?
Am I supposed to use another specific example or settings for Attiny2313?

From latest commits and docs, I thought Attiny2313 is supported.
What should I do to be able to compile the sample?

Thanks

2bndy5, 2021-01-20:
>The compiled program is 2816 bytes which exceeds available 2048 bytes with LTO enabled

This should answer your question. There simply isn't enough memory in the ATTiny2313 for that example. Although, I don't know what "with LTO enabled" means... 

looking at [the CI for that example (compiled for the ATTinyx313 chips) upon last push to master](https://github.com/nRF24/RF24/runs/1578955131?check_suite_focus=true#step:3:19) branch: I see some warnings that should be addressed.

ceyhunozgun, 2021-01-21:
Hi @2bndy5 , thanks for your reply.

I have represented rf24ping85 as a sample because I think it is the minimal working example that don't include any unnecessary logic (also it doesnt include any code that especially specific to Attiny85) that would cause any increase in the total compiled code size.
Also I have tried the code original poster sent which I think it also minimum code to work with nRF24 library.
It also exceeds the total memory size of Attiny2313a which is 2048 bytes.

I don't know what else the minimal code would be that would work with Attiny2313 then.
That is why I was asking whether there is an option, settings or other minimal code sample that I should use to be able to use nRF24 library with Attiny2313.

If not, how can we say Attiny2313 is supported?
Is there anything I miss?
If it is, would you please direct me?

Thanks

2bndy5, 2021-01-21:
If you take out all the `Serial` stuff from the *gettingStarted.ino* example (the most minimal code example for this library), and it still exceeds the 2048 max memory, then we might have to call the ATTiny2313 unsupported. If you do get an example to fit on the ATTiny2313, then you may not have a lot of memory left to use the library for whatever your project is.

2bndy5, 2021-01-21:
@ceyhunozgun  you got e curious, so I tried to compile a modified gettingStarted.ino example (with all `Serial` stuff commented out, and it still doesn't fit.

>Sketch uses 2868 bytes (140%) of program storage space. Maximum is 2048 bytes.
Global variables use 95 bytes (74%) of dynamic memory, leaving 33 bytes for local variables. Maximum is 128 bytes.

And when I defined `MINIMAL` like so
```cpp
#include <SPI.h>
#define MINIMAL
// #include "printf.h"
#include "RF24.h"
```

the compile size is still too large
>Sketch uses 2408 bytes (117%) of program storage space. Maximum is 2048 bytes
Global variables use 59 bytes (46%) of dynamic memory, leaving 69 bytes for local variables. Maximum is 128 bytes.

@TMRh20 we may have to call the ATTiny2313 unsupported due to lack of sufficient memory capacity.
@ceyhunozgun You might find another library for the nRF24L01 _might_ work for the ATTiny2313 (maybe try the [nrflite llibrary](https://github.com/dparson55/NRFLite) made for the ATTiny84/85), but we cannot help you with other libraries. If you do find a library that works on the ATTiny2313, please leave us a comment on this thread. We'd be happy to include a link in the docs for such a library.

TMRh20, 2021-01-22:
@2bndy5 Yeah I think so. Its only 2Kb MCU so not much can be done.

2bndy5, 2021-01-22:
I agree. A lighter version of this library would not be possible for the ATTiny2313.

Compiling the [ATTinyCore's Digital Potentiometer example](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/SPI/examples/DigitalPotControl/DigitalPotControl.ino) (a very basic program that uses SPI) occupies almost half the storage memory on a ATTiny2313
>Sketch uses 948 bytes (46%) of program storage space. Maximum is 2048 bytes.
Global variables use 14 bytes (10%) of dynamic memory, leaving 114 bytes for local variables. Maximum is 128

ceyhunozgun, 2021-01-27:
@2bndy5 thanks for the confirmation. Maybe changing the docs about the situation can make the job of future researchers easier, so they won't go into trouble trying.

For the record, I was able to fit a simple program into Attiny2313 (2024 bytes) with the use of a trimmed version of [NRFLite](https://github.com/dparson55/NRFLite) library (tx side only, not rx side) only by trimming some unnecessary methods like printDetails, etc. I mean I didn't have to code my own SPI code etc. Although almost nothing left to my app for code space, for a simple remote control project with an Arduiono joystick module which doesn't require a library, it does the job.

Regards

PR #737 (2021-02-27): conclude work on a small branch
This PR concludes small changes on my patch-ping85-example branch. (forgot I started this branch 😉 )

- removes a unused variable (which avoids a warning during compilation) from ping85 example
- add a warning about ATTiny2313 unsupported to ATTiny's "Related Pages" and the "main Page"'s footnote (under the pinout table). This closes #527 
- adds a comment to identify what a seemingly rogue `#endif` pertains to
- adds useful troubleshooting info about using PA/LNA modules in COMMON_ISSUES.md (includes bonified pictures of ghetto_shielding with foil & electrical  [... cut]
closed 2021-02-27 by 2bndy5