nRF24/RF24 #529: Error compiling simple test on NodeMCU 1.0 with nRF24L01

mRudzki, 2019-10-03:
Hello, I'm new to RF24, and when writing some simple test for my ESP8266 I got strange error.
```
In file included from C:\Users\Origin\Documents\Arduino\libraries\RF24-master\RF24.cpp:9:0:

C:\Users\Origin\Documents\Arduino\libraries\RF24-master\nRF24L01.h:33:21: error: expected unqualified-id before numeric constant

 #define STATUS      0x07

                     ^

C:\Users\Origin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/sdk/include/c_types.h:76:3: note: in expansion of macro 'STATUS'

 } STATUS;

   ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```
I have basically no idea, if it is my fault or just esp library and RF24 library are having hard time with each other.

wmarkow, 2019-10-04:
There is a conflict between two different **STATUS** definitions:
 * one comes from nRF24L01.h:33:21 ```#define STATUS      0x07```
 * the other comes from esp8266

The issue has been already fixed in cd52d61bc08 commit as suggested in #125. You are using old version of RF24 library. Please use the latest one.

mRudzki, 2019-10-05:
Thanks for your answer.

closed 2020-01-28 by Avamander