nRF24/RF24 #531: terminate called after throwing an instance of 'int'

TheMlok, 2019-10-04:
Hello,

when I compile the examples on Raspberry Pi 2 model B v1.1 I usually got this error. When I debug the code, method radio.available() returns -1 instead of bool.

I have working SPI 

spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

00 00 00 00 FF FF
C0 00 00 00 00 00
00 00 00 FF FF 00
00 00 00 00 00 FF
FF 00 00 00 00 00
3F 80 00 1F 80 00
FF FC

Running out of ideas.

Thanks

M.

wmarkow, 2019-10-05:
According to the latest code of RF24 library, the ```available()``` method should return either **0** or **1** which is then casted to ```bool```. Why do you see **-1**?

TheMlok, 2019-10-05:
I am using latest version, compiled from github, no errors or warnings during compile. I have not debugged deeply yet, lack of time :( I will do it.
According this documentation
http://tmrh20.github.io/RF24/classRF24.html#a127105eb7a3b351cfe777c1cec50627a
it is 0 or 1, I think that -1 is made by some kind of unhandled exception deeper. But I don't know why yet.

agrim12345, 2019-10-09:
whats the solution?

CoRfr, 2020-01-26:
I'm also getting this error, but it happens in RF24::begin(), on:
```
write_register(NRF_CONFIG, 0x0C);
```

```
write_register(00,0c)
terminate called after throwing an instance of 'int'
```

CoRfr, 2020-01-26:
Arg I feel a bit stupid, there is a check to make sure that the process runs as root in `utility/RPi/spi.cpp`:
```
void SPI::beginTransaction(SPISettings settings)
{
    if (geteuid() != 0) {
        throw -1;
    }
```
`throw -1` is however not the most obvious error code.

closed 2020-03-29 by TMRh20