nRF24/RF24 #532: Unable to communicate between Raspberry Pi 3B+ and Raspberry Pi 0

ashish-k-m, 2019-10-09:
Hello guys,

I have made the correct wiring between the Raspberry Pi's and the NRF24+PA+LNA.  I have also connected a 100 uF capacitor between V and GND and also an external 9V power supply but whenever Irun the gettingstarted program and I select ping out I get the error "Failed, response timed out". I am posting screenshots of the terminal below. Please tell me where I am going wrong and if there is any possible way to fix it
[image]
[image]

githubforsank, 2019-10-17:
Same issue here .. i opened the issue too and waiting for someone to address.
Garbage null values with SPIDEV, RPi, wiringPi drivers

Avamander, 2019-10-17:
The way to address this issue is to eliminate one-by-one in which component the issue lies in. As usual with debugging. It's also worth reading other issues about your setup, in many cases people find mistakes in their setups and then vanish. 

0. Check that you have a smooth power supply, 100uF is not nearly enough with most 3.3V regulators.
1. Check wiring, are you using the right SPI ports on both rPis?
2. Check if SPI is working, if you change radio speeds, does that change appear in `printDetails()`?
3. Check if rPi is causing issues, do the radios work with two Arduinos?

Do those steps first please and then we can investigate further.

closed 2020-03-29 by TMRh20