nRF24/RF24 #535: RPI3B+ NRF24L01+  random communication  behaviour

githubforsank, 2019-10-20:
Hi Contributors,
This is my 5th day into the unforeseen communication issue of my B+ and nrf24 module. I am building a prototype which requires the beautiful library RF24/RF24Network and Mesh. Mesh network is my utmost requirement and no other library provides as this one. I have been trying to communicate an RPI B+ with a nRF as master node and Arduinos as slave nodes. My RPI is unable to read address bytes of the module but I have some weird observations which i am noting down. I have tried all the myriad of things available in the internet and still unable to get an answer. 

Informations:
OS : Raspbian ( Buster )

Observations:
1- The address bytes changes after multiple start of the gettingstarted 
[image]
The 00's changes to FF's and 22's sometimes.
2- Pong back role always replies back with mostly 0 but sometimes random values.
[image]
3- Pingout role gets 0 mostly as response but sometimes random values as 128
[image]
4- Taking a wire out is detected by the library and connection error is thrown in the terminal.

Steps Tried:
1- 100th checked the connection ( using multimeter )
2- Changed driver to SPIDEV, wiringPi, MRAA, RPi and pydev
3- Read somewhere the bcm driver in the newer kernel might be broken. So updated bcm2835.c.
4- I couldn't find a similar problem of other friends around in the internet  ( although i went very close )  so downgraded OS and Kernel to (4.1) VERSION="8 (jessie)" ID=raspbian. ( Release around July , 2016) 
5- Tried the python library and pingpair.py and got the same results. 
6- Tried other nr24 standalone cpp and python libraries. Read somewhere changing spi.speed = 50000 max.
7- SPI Loopback test worked with pigpoid and spidev_test.cpp ( device : /dev/spidev.0.0).
8- piscope LIVE analysis: when running ./gettingstarted 
The MOSI was always bursting data.
The SCLK burst after 3-4 seconds. 
The MISO was sending random bursts for which we were receiving 0 and garbage ( I assume ) 
but still I wonder how loopback was working which proper signal in SPI pins. 
I was trying to isolate this as a hardware and software issue. 
9- Took 2 arduino Nano's and tested all my 3 nrfmodules . They all work and mesh network got created with one arduino as a master node. Used all the modules in the RPi none worked. 
10- Tried changing to other GPIO pins for CE and tried changing bus speed from 8MHz, 4 MHz, 1 etc.
11- All these steps on the all the OSes.

Assumptions:
1- Checked noise and put 500/1000 uf caps over the vcc and gnd.
2- 3.3 v regulated power supply to the module. ( From Pi & 3.3 regulator)

It's driving me crazy please help !!. 
PS: I don't have a second box but I have ordered a RPI Zero W and waiting to test on it.

Avamander, 2019-10-20:
The status output indicates incorrect wiring/lack of communication. Mkae
sure you're using the right SPI port on your RPi.

githubforsank, 2019-10-20:
I am using the spi 0 . 19 21 23 mosi miso clk 24 for csn and 15 ce. I have cross checked nrf device multiple times . Can you help me please. I can post pictures of connection if you want. I have checked connectivity through multimeter too. Unable to figure out the problem

Avamander, 2019-10-25:
You can try decoupling the 3.3V and GND pins on the nRF24 itself with an electrolytic capacitor and see if that helps. But if the wiring is correct then your RPi's SPI is either not enabled or does not work for some other reason you have to investigate.

closed 2020-01-28 by Avamander