nRF24/RF24 #536: I don't know how to connect the wires

adrianTNT, 2019-10-21:
Hello.
I use an Arduino Nano. The pins/wiring instructions confuse me ...
In the GettinStarted example it just says: 
```
"/* Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 7 & 8 */
RF24 radio(7,8);"
```

What does it mean exactly ? Digital pins 7 and 8 from the arduino go in which of the 3 SPI pins on the transceiver ?

TMRh20, 2019-10-22:
See https://www.deviceplus.com/how-tos/arduino-guide/nrf24l01-rf-module-tutorial/ or other tutorials on how to connect your radios. Pins 7 and 8 are the default CE/CS pins.

adrianTNT, 2019-10-22:
Thanks, that helps because it has the images of wiring.

But ... isn't it better for the examples to include some wiring details ? 
I am a beginner with this and the part I mentioned in the initial post made no sense for me. 

**SPI bus plus pins 7 & 8** , that is all it says (out of the 8 radio module pins) and even these two were not clear at that time where to connect them.

I was able to make it work kind of fast using a simple_tx and simple_rx sample (similar to the tutorial you mentioned), but I think the examples/documentation can be better.

Avamander, 2019-10-25:
It's less error-prone to say in the documentation to connect `the thing` to the SPI bus rather than some `x` ,`y`, `z` and `q` pins which might vary from internal pin number to what's written on the board, creating only confusion. There's also a tremendous amount of tutorials and schematics online for every imaginable board, it's wasting effort to just repeat that information. If you Google `your board name` + `SPI bus` then you do get a lot of images where the SPI bus is on your specific board. You will probably also encounter people talking about `I2C bus` in the future, you can find where that bus is exactly the same way.

closed 2019-10-25 by Avamander