nRF24/RF24 #541: digitalWrite(LED_BUILTIN, HIGH); Not working after radio.begin();

alkuentrus, 2019-11-01:
Hi, I a using an Arduino Nano and IDE 1.8.10. The code attached works fine for another aspects, not for Digital Out for Led pin. If I comment the code radio.begin(); It works fine.... Can you help me?
`  #include <RF24.h>
  #include <SPI.h>
  #include <RF24Audio.h>
  #include "printf.h"  
      
  RF24 radio(7,8);    // Set radio up using pins 7 (CE) 8 (CS)
  RF24Audio rfAudio(radio,0); // Set up the audio using the radio, and set to radio number 0

void setup() {
 Serial.begin(115200);
 printf_begin();
//radio.begin();
 pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, HIGH);
}

void loop() {

}`

TMRh20, 2019-11-02:
SPI uses pin 13, so you can’t really use the built in LED while using SPI.

alkuentrus, 2019-11-02:
Oh my, oh my!!! Did'n remember that!!! Sorry :(

olback, 2019-11-08:
@alkuentrus 

You should close this issue now that you know what's up so that when other people search for the same issue, they know it's resolved. :+1:

alkuentrus, 2019-11-08:
@olback , You are right! Id didn't realize I had to :) Going to do so, thks!

closed 2019-11-08 by alkuentrus