nRF24/RF24 #545: Makefile.inc Not Created for "make install"

harperalmo, 2019-11-15:
I cloned the RF24 material onto my Raspberry pi 4 using

    git clone https://github.com/nRF24/RF24.git

then moved to the created RF24 directory, and following directions found at 

       http://tmrh20.github.io/RF24/RPi.html

 invoked make like this:

    sudo make install -B

The output from make was:

   [Running configure]
   make: execvp: ./configure: permission denied
   Makefile:17: Makefile.inc: No such file or directory
   make:  ***[Makefile:71: Makefile.inc] Error127

   

It appears that the configure file (a Bash shell script) should build Makefile.inc, but it appears that the script cannot run due to permissions. Permissions for configure are:    -rw-r--r--
I tried adding permissions using sudo chmod +w and sudo chmod +x to see if that would help, but
the permissions did not change. I'll have to look into that. Can anyone see reasons why the "make install -B" is failing?

Also, it appears that the BCM2711 (the Rpi4 gpu) is not part of the supported hardware for this library, so perhaps my initial question is moot. Does anyone know if there are plans for 2711 support?

TMRh20, 2019-11-17:
Not sure why the problems with make/configure but there is a hack to support RPI4 with the bcm driver. No plans here to support 2711 specifically.

fredo838, 2019-12-14:
Have you tried `sudo ./configure` ?

closed 2020-08-11 by TMRh20