nRF24/RF24 #546: getting errors during sudo make install -B

adrianTNT, 2019-11-21:
I am trying to install the nRF24 library on raspberry pi zero w.
Based on [thse instructions](http://tmrh20.github.io/RF24/RPi.html), I did

```
sudo apt-get update 
sudo apt-get upgrade
```
Next, I understand that I need to configure **one** of these 3 ? BCM2835 / MRAA / SPIDEV 
On the build step I just ran this to use SPIDEV:
```
./configure --driver=SPIDEV
sudo make install -B
``` 
Do I need to also install/configure BCM2835 and MRAA ?
Because the above command returned these errors:

```
sudo make install -B
[Running configure]
[SECTION] Detecting arm compilation environment.
  [OK] arm-linux-gnueabihf-gcc detected.
  [OK] arm-linux-gnueabihf-g++ detected.
[SECTION] Detecting target machine.
./configure: line 66: warning: command substitution: ignored null byte in input
[OK] machine detected: SoC=BCM2835, Type=RPi, CPU=armv6l.
[SECTION] Detecting DRIVER
  [OK] DRIVER detected:RPi.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[OK] Finished.
arm-linux-gnueabihf-g++ -fPIC -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
arm-linux-gnueabihf-g++ -fPIC -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/RPi/spi.cpp
arm-linux-gnueabihf-gcc -fPIC -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/RPi/bcm2835.c
utility/RPi/bcm2835.c: In function ‘bcm2835_gpio_pad’:
utility/RPi/bcm2835.c:491:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
   if (bcm2835_pads == MAP_FAILED)
   ^~
utility/RPi/bcm2835.c:494:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
     volatile uint32_t* paddr = bcm2835_pads + BCM2835_PADS_GPIO_0_27/4 + group;
     ^~~~~~~~
utility/RPi/bcm2835.c: In function ‘bcm2835_gpio_set_pad’:
utility/RPi/bcm2835.c:504:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
   if (bcm2835_pads == MAP_FAILED)
   ^~
utility/RPi/bcm2835.c:507:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
     volatile uint32_t* paddr = bcm2835_pads + BCM2835_PADS_GPIO_0_27/4 + group;
     ^~~~~~~~
arm-linux-gnueabihf-g++ -fPIC -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/RPi/interrupt.c
[Linking]
arm-linux-gnueabihf-gcc -shared -Wl,-soname,librf24.so.1 -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -o librf24.so.1.3.3 RF24.o spi.o bcm2835.o interrupt.o -pthread
[Installing Libs to /usr/local/lib]
[Installing Headers to /usr/local/include/RF24]
```

Avamander, 2019-11-21:
Those are warnings.

closed 2019-11-21 by Avamander

TMRh20, 2019-11-21:
As an aside, don’t use the -B switch on make install after running configure, because it re-runs configure with the default options.

adrianTNT, 2019-11-21:
Thanks.
I did that without the `-B` and the warnings were gone.
```
sudo ./configure --driver=SPIDEV
sudo make install
```
Maybe you want to remove the `-B` from here: http://tmrh20.github.io/RF24/RPi.html 
It is after each configure command with `--driver=`

TMRh20, 2019-11-21:
Thanks for letting me know it’s wrong in the docs. Will get that fixed soon.