nRF24/RF24 #548: ‘RPI_V2_GPIO_P1_15’ was not declared in this scope

adrianTNT, 2019-12-02:
I am sending data from arduino with a simple short code...
(A similar code receives data OK with arduino).

```
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

// RADIO pins
RF24 radio(7, 8); // CE, CSN

const byte address[6] = "00333";

void setup() {
  radio.begin();
  radio.openWritingPipe(address);
  radio.setPALevel(RF24_PA_LOW);
  radio.stopListening();
}

void loop() {
  const char text[] = "Hi ";
  radio.write(&text, sizeof(text));
}
```
But what I need now is to receive **on raspberry pi zero**
Can someone tell me what code to use on raspberry pi side given the above send method ?

I am getting this error when compiling a code with `make` (deviated from `examples_linux/gettingstarted.cpp`):
```
error: ‘RPI_V2_GPIO_P1_15’ was not declared in this scope
 RF24 radio(RPI_V2_GPIO_P1_15, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);
            ^~~~~~~~~~~~~~~~~
```

That config looks like this, derivated from gettingstarted file:
```
/****************** Raspberry Pi ***********************/

// Radio CE Pin, CSN Pin, SPI Speed
// See http://www.airspayce.com/mikem/bcm2835/group__constants.html#ga63c029bd6500167152db4e57736d0939 and the related enumerations for pin information.

// Setup for GPIO 15 CE and CE0 CSN with SPI Speed @ 8Mhz
RF24 radio(RPI_V2_GPIO_P1_15, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);

//RPi Alternate, with SPIDEV - Note: Edit RF24/arch/BBB/spi.cpp and  set 'this->device = "/dev/spidev0.0";;' or as listed in /dev
// RF24 radio(22,0);
```

I assumed that is the right config method to enable, I am using the pins like in attached picture (**using pins 15 and 24**).

What am I doing wrong ? 
And if I enable that SPIDEV method next to this one, do I need to switch pins/wires ?

[image]

TMRh20, 2019-12-02:
Are you sure you are not already using the SPIDEV driver? You shouldn’t get this error if using the BCM driver.

In either case you can just use RF24 radio(22,0); for gpio22 and CS 0 with your current pin configuration.

adrianTNT, 2019-12-03:
I uncommented the line with `RF24 radio(22,0);` and it worked OK. Thanks.

Now it says it receives the data (sent by my arduino), but I can't extract the actual text. I find the code in `gettingstarted.cpp` too complex to be able to create a simple "receive only" code from it. 

If is not too much to ask, could you tell me how to translate this arduino receiver into a .cpp like the `examples_linux/gettingstarted.cpp` ? That would save me a few days of guessing and trying.

```
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

RF24 radio(7, 8); // CE, CSN

const byte address[6] = "00333";

void setup() {
  
  Serial.begin(115200);
  radio.begin();
  radio.openReadingPipe(0, address);
  radio.setPALevel(RF24_PA_LOW);
  radio.startListening();
  
}
void loop() {
  if (radio.available()) {
    char text[32] = "";
    radio.read(&text, sizeof(text));
    Serial.println("Received: "+(String)text);
  }
}
```

closed 2020-03-29 by TMRh20