nRF24/RF24 #550: NRF24 + ATMEGA4809 (incompatible to megaavr)

alban-d-b, 2019-12-22:
Hey :)
I tried to compile this library working for an ATMEGA4809 chip, because it seems like the default ATmega328p chip just hasn't enough SRAM for my needs. But the compiler just says that the library isn't compatible to megaavr chips, like the ATMEGA4809. Is there nevertheless a possibility to get this working on a this chip?

error report:
```
Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino Uno WiFi Rev2, None (ATMEGA4809)"

Build-Optionen wurden verändert, alles wird neu kompiliert
WARNUNG: Bibliothek RF24 behauptet auf (avr, arm, x86, esp8266, esp32) Architektur(en) ausgeführt werden zu können und ist möglicherweise inkompatibel mit Ihrem derzeitigen Board, welches auf (megaavr) Architektur(en) ausgeführt wird.
C:\Users\*\Documents\Arduino\libraries\RF24\RF24.cpp: In member function 'void RF24::csn(bool)':

C:\Users\*\Documents\Arduino\libraries\RF24\RF24.cpp:69:27: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

  digitalWrite(csn_pin,mode);

                           ^

C:\Users\*\Documents\Arduino\libraries\RF24\RF24.cpp: In member function 'void RF24::ce(bool)':

C:\Users\*\Documents\Arduino\libraries\RF24\RF24.cpp:80:51: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

   if (ce_pin != csn_pin) digitalWrite(ce_pin,level);

                                                   ^

exit status 1
Fehler beim Kompilieren für das Board Arduino Uno WiFi Rev2.

```

Thanks for any help :D

Avamander, 2019-12-22:
Arduino IDE supports adding a list of compatible archidectures to a library, `megaavr` hasn't been listed as a compatible architecture because it hasn't been tested. In the end it's a warning. However in this case it seems that there are actual incompatibilities. For some reason your board doesn't take boolean as a `digitalWrite` parameter, is this on an officially supported Arduino hardware?

Avamander, 2019-12-22:
I'm not sure if this would work, but maybe it can be overloaded (I can't test right now):
```
void digitalWrite(pin_size_t pin, bool value) {
    if (value) {
        digitalWrite(pin, HIGH);
    } else {
        digitalWrite(pin, LOW);
    }
}
```

Add it before the `#include`-s.

alban-d-b, 2019-12-24:
Thanks for your help :) At the time I develop on an Arduino Mega (ATMEGA2560). But later on I wanted to solder everything on a circuit board. As the ATMEGA2560 is only available as SMD chip, i tried to compile the code for the Arduino Uno WiFi Rev2 (ATMEGA4809), which ran into the known error above. I also tried to compile it for the ATmega644pa with this library (https://github.com/MCUdude/MightyCore), which seemed to work. So I will probably try my best with this one, before changing anything in the nRF24 library.

Avamander, 2019-12-25:
You can add that function to your sketch before `#include` directives, it doesn't require modifying RF24.

chell42, 2020-02-07:
hi i am also trying to use a A4809 but this time in the nano every
i have added the code fix from avamander's above comment
this has fixed on issue but i have more to add
please see below the error message my ide is spitting out

`Arduino: 1.8.11 (Windows 10), Board: "Arduino Nano Every, None (ATMEGA4809)"

WARNING: library RF24-master claims to run on avr, arm, x86, esp8266, esp32 architecture(s) and may be incompatible with your current board which runs on megaavr architecture(s).
C:\Users\unusual\Documents\Arduino\libraries\RF24-master\RF24.cpp: In member function 'void RF24::csn(bool)':

C:\Users\unusual\Documents\Arduino\libraries\RF24-master\RF24.cpp:68:31: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

     digitalWrite(csn_pin, mode);

                               ^

C:\Users\unusual\Documents\Arduino\libraries\RF24-master\RF24.cpp: In member function 'void RF24::ce(bool)':

C:\Users\unusual\Documents\Arduino\libraries\RF24-master\RF24.cpp:79:35: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

         digitalWrite(ce_pin, level);

                                   ^

exit status 1
Error compiling for board Arduino Nano Every.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
i am trying to test my NRF24L01+ for coms between arduino and radio.
i am using the following code from a fourom

`void digitalWrite(pin_size_t pin, bool value) {
    if (value) {
        digitalWrite(pin, HIGH);
    } else {
        digitalWrite(pin, LOW);
    }
}
#include <SPI.h>
#include <nRF24L01.h>   //nRF2401 libarary found at https://github.com/tmrh20/RF24/
#include <RF24.h>          //nRF2401 libarary found at https://github.com/tmrh20/RF24/
#include "printf.h"

const int pinCE = 7;
const int pinCSN = 8;

RF24 wirelessSPI(pinCE, pinCSN);
 
const uint64_t pAddress = 0xB00B1E5000LL;           

void setup() 
{
  Serial.begin(9600);   
  printf_begin();       
  wirelessSPI.begin();         
  wirelessSPI.setAutoAck(1);         
  wirelessSPI.enableAckPayload();             
  wirelessSPI.setRetries(5,15);                 
  wirelessSPI.openWritingPipe(pAddress);       
  wirelessSPI.stopListening();
  wirelessSPI.printDetails();                 

}

void loop() 
{ 
 
 
}`

i hope tis helps get the libary working

Avamander, 2020-02-15:
Fixed with https://github.com/nRF24/RF24/commit/9f55c0c577b229f553bca4c73e3045cc89bf73c1

closed 2020-02-15 by Avamander