nRF24/RF24 #552: Error compiling with ESP32 + NRF24L01+

slowgergo, 2019-12-29:
I tried compiling several codes for my ESP32 devkit v4 but all of them gave the same error :

```
F:\Projects\Arduino\libraries\RF24\RF24.cpp:579:101: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

   printf_P(PSTR("Data Rate\t = " PRIPSTR "\r\n"),pgm_read_ptr(&rf24_datarate_e_str_P[getDataRate()]));
```

It gave 4 errors like this for all 4 lines 579-582 in RF24.cpp

The strange thing is that i even tried with with an empty code just adding the defines on top 
```
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <printf.h>
```
And even like this it gave the errors.

I tried the same codes with my Arduino Micro and it worked without errors.

I have been looking around for 3 hours but I'm clueless.  I saw some things to change here and there in RF24_config.h, Rf24.cpp but everything they suggested were already there so i didn't modify anything. I reinstalled the library, the board but nothing changed.

Any help would be appreciated. 

The full error:
```
Build options changed, rebuilding all
F:\Projects\Arduino\libraries\RF24\RF24.cpp: In member function 'void RF24::printDetails()':

F:\Projects\Arduino\libraries\RF24\RF24.cpp:579:101: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

   printf_P(PSTR("Data Rate\t = " PRIPSTR "\r\n"),pgm_read_ptr(&rf24_datarate_e_str_P[getDataRate()]));

                                                                                                     ^

F:\Projects\Arduino\libraries\RF24\RF24.cpp:579:101: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

F:\Projects\Arduino\libraries\RF24\RF24.cpp:580:95: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

   printf_P(PSTR("Model\t\t = " PRIPSTR "\r\n"),pgm_read_ptr(&rf24_model_e_str_P[isPVariant()]));

                                                                                               ^

F:\Projects\Arduino\libraries\RF24\RF24.cpp:580:95: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

F:\Projects\Arduino\libraries\RF24\RF24.cpp:581:104: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

   printf_P(PSTR("CRC Length\t = " PRIPSTR "\r\n"),pgm_read_ptr(&rf24_crclength_e_str_P[getCRCLength()]));

                                                                                                        ^

F:\Projects\Arduino\libraries\RF24\RF24.cpp:581:104: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

F:\Projects\Arduino\libraries\RF24\RF24.cpp:582:99: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

   printf_P(PSTR("PA Power\t = " PRIPSTR "\r\n"),  pgm_read_ptr(&rf24_pa_dbm_e_str_P[getPALevel()]));

                                                                                                   ^

F:\Projects\Arduino\libraries\RF24\RF24.cpp:582:99: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]

cc1plus.exe: some warnings being treated as errors

exit status 1
Error compiling for board ESP32 Dev Module.
```

slowgergo, 2019-12-29:
I just tried with my ESP8266 chip and that worked too without errors but this is reserved for another project :(

Avamander, 2019-12-29:
Remove `-Werror` or `-Werror=format=` from your compilation flags and try again, your environment is treating a warning as an error and aborting.

slowgergo, 2019-12-29:
How exactly can i do that? Im using arduino IDE. Should i use something else?

Avamander, 2019-12-29:
There might be an option in board selection menu for that. You can also just not use the `printDetails()` and `printf.h` library and it should be fine.

slowgergo, 2019-12-29:
I tried to compile your/the library's getting started code and it gave the same error and there isn't any `printDetails()` nor the `printf.h` used.

slowgergo, 2019-12-29:
Okay found how to remove -werror flags and it compiled. :)

Ill try actually uploading and testing

closed 2019-12-29 by slowgergo