nRF24/RF24 #554: Struct transmission stopped working after adding variables

md88x, 2020-01-12:
Dear all,

I have a small weather station project with an atmega standalone+nrf24  outside transmitting the sensor data to an esp8266+nrf24 indoor.
Everithing is working like a charm using this struct :

struct package
  {
    float t = 0;
    float h = 0;
    float p = 0;
    int raincounter = 0;
    };
struct package data;

I wanted to simply added to the transimtter and the receiver this simply byte flag variable and the receiver went bananas displaying one random big number:

struct package
  {
    byte flag = 0;
    float t = 0;
    float h = 0;
    float p = 0;
    int raincounter = 0;
    };
struct package data;

The payload should be 1+4+4+4+2 = 15 bytes, well under the 32 limit.

The sketch regarding the trasmission in very simple:

Tx setup:
radio.begin();
radio.openWritingPipe(address);
radio.stopListening();

Tx in the loop section is:

radio.write(&data, sizeof(data));
and then goes to deep sleep until the next sensor reading and trasmission.

The Receiver setup:
radio.begin();
radio.openReadingPipe(0, address); 
radio.startListening();

loop:
while (radio.available())
    radio.read( &data, sizeof(data) ); 
and then display the data on a tft and uploads on thingspeak.

I am studying again the examples but I am not understaning what is going wrong.

Thanks in advance for the help.

TMRh20, 2020-01-13:
To start, you can compare the size of the struct on the different devices, and you should notice a difference.

This is because the atmega is 8-bit while esp is 32 bit.

Basically the esp will store the data in chunks of 4 bytes, so padding will be added in cases where you have for example, a byte followed by a float in your structure. In that case a padding of 3 empty bytes will be added after the byte, and before the float. 

In your case, for simplicity’s sake, I would move the byte to the end of the data structure, and add a second byte, so the size of the data structure is the same on both devices.

From there you can modify as desired, to understand how data is stored/packed in memory.

wmarkow, 2020-01-13:
As @TMRh20  has commented, the struct packing has a meaning. You can also tell the compiler to not to pack the struct, then the additional padding will not be added.

Take a look at https://github.com/esp8266/Arduino/issues/1825 for some details.

Another thing is the size of the data types between two platforms. On Arduino Uno the **byte** is 1 byte length (https://www.arduino.cc/reference/tr/language/variables/data-types/byte/). It looks like in ESP8266 the **byte** is also 1 byte long (https://www.arduinoforum.de/arduino-Thread-L%C3%A4nge-der-Datentypen-beim-ESP8266). So in the case of **byte** data this doesn't affect the length of the structure, so you need to take a look at the padding, as @TMRh20 has noticed.

But if it goes about **int** variable, the we have:
 * 2 bytes length in Arduino UNO (AVR)
 * 4 bytes length in ESP8266

So when sending the struct with **int** inside, you may hit also the problem of data lenght (not only the packing of the structure). In this case it is better to use the data types like **uint8_t**, **uint16_t** and so on.

md88x, 2020-01-13:
Thank oyu all, your answers gave me something to tink about.

From what I am understandin in general is better to declare variables as uintXX_t and to avoid data structure alignement from the compiler and to save a little space of the struct i should declare the struct as follows:

```
struct __attribute((__packed__)) package
  {
    uint8_t flag = 0;
    float t = 0;
    float h = 0;
    float p = 0;
    uint32_t raincounter = 0;
    };
struct package data;
```
Is that correct?
This week I am away from my project, I will test it this weed end.

Thanks

md88x, 2020-01-18:
Dear all I've tested the code and now everithing works fine.

Thankyou again for the help

closed 2020-01-18 by md88x