nRF24/RF24 #555: Static payload always availible

bakerw71, 2020-01-12:
Regardless of the status of the transmitter (even turned off) the module is reporting available packets.
Reading these packets into the Data structure, just a simple structure to contain a few 16bit values results in filling the structure with 65535 most of the time, but also random values like 57727 and 65407. I have triple checked the wring which seems okay. I also have a LoRa SX1278 chip but that should be silenced by writing its CS pin HIGH.

I am using an ESP8266 module programmed with an FTDI breakout.
And a NRF24L01 module. I have tried using a 4.7uF electrolytic capacitor to smooth input voltage.
Uploading using visual micro.
Board: "Generic ESP8266 Module"

```
#define LORACS 15
#define NRFCS 4
#define NRFCE 16
RF24 nrf(NRFCE, NRFCE); //ce pin 16 isn ideal
void setup() {
                digitalWrite(LORACS, HIGH);
 NRFSetup:             nrf.begin();
		nrf.setAutoAck(false);
		nrf.setCRCLength(RF24_CRC_8);
		nrf.setDataRate(RF24_250KBPS);
		nrf.setChannel(108);
		nrf.setPALevel(RF24_PA_MAX);
		nrf.disableDynamicPayloads();
		nrf.setPayloadSize(sizeof(Data));
		nrf.setRetries(0, 0);
		nrf.csDelay = 1;
                if (nrf.getCRCLength() != RF24_CRC_8) {
			Serial.println("CRC Incorrect");
			goto NRFSetup;
		}

		nrf.openReadingPipe(0, NRFADDRESS);
		nrf.startListening();
                if (nrf.failureDetected) goto NRFSetup;
}

void loop(){
        if (nrf.available() && (nrf.getPayloadSize() == sizeof(Data))) {
		nrf.read(&dataNRF, sizeof(Data));
		Serial.print(dataNRF.Throttle);
	}
}
```

[image]

Avamander, 2020-01-12:
Hi, just to make sure this isn't your code, can you upload one of the PingPair examples onto your board and see if you see random packets available the same way?

bakerw71, 2020-01-12:
Oh my, just saw this:
`RF24 nrf(NRFCE, NRFCE); `
The second param should be NRFCS
I don't understand how the setup executed correctly, especially since I set all the CS pins high in my tests.

closed 2020-01-28 by Avamander