nRF24/RF24 #556: cepin / cspin inverted?

cmburcus, 2020-01-17:
Something doesn't add up here. I was losing my mind for hours until I realized why my module was doing weird things.

According to the [documentation](http://tmrh20.github.io/RF24/classRF24.html), this is the constructor:
`RF24 (uint16_t _cepin, uint16_t _cspin)`

In my code, I went ahead and defined the following:
```
#define CE_PIN 7
#define CSN_PIN 8
RF24 radio(CE_PIN, CSN_PIN);
```

I made sure the CE pin is really connected to the digital pin 7 and the CSN pin is really connected to the digital pin 8. (I think I checked it over 10 times thinking I was crazy). I unplugged, re-plugged many times rotating arduinos and modules (i have 3 arduinos and 3 modules).

It would appear the constructor actually takes CSN pin as the first argument and the CE pin as the second? Because if I switch them, it works... any help or confirmation would be highly appreciated.

Thank you.

Avamander, 2020-01-17:
Your platform or the board you're using is wonky, what exactly are you using?

cmburcus, 2020-01-17:
Arduino Mega2560 x1 and Arduino Uno x2
Modules are nRF24L01+

I must be going crazy tho because now it works the correct way. I was sure the pins were inverted but... who knows anymore. If it happens again I'll re-open and take pics / not fiddle anymore with it

closed 2020-01-17 by cmburcus