nRF24/RF24 #561: RF24 Library with Arduino Uno WiFi R2

clearlincognito, 2020-01-26:
I am getting this error every time I try to compile it. Is there any way to fix this? Thank you.
_`Arduino: 1.8.10 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

/private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/hardware -hardware /Users/clearlincognito/Library/Arduino15/packages -tools /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/clearlincognito/Library/Arduino15/packages -built-in-libraries /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/libraries -libraries /Users/clearlincognito/Documents/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=on -vid-pid=0X03EB_0X2145 -ide-version=10810 -build-path /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216 -warnings=none -build-cache /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_cache_692692 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /Users/clearlincognito/Documents/Arduino/sketch_jan04a/sketch_jan04a.ino
/private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/hardware -hardware /Users/clearlincognito/Library/Arduino15/packages -tools /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/clearlincognito/Library/Arduino15/packages -built-in-libraries /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/libraries -libraries /Users/clearlincognito/Documents/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=on -vid-pid=0X03EB_0X2145 -ide-version=10810 -build-path /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216 -warnings=none -build-cache /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_cache_692692 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /Users/clearlincognito/Documents/Arduino/sketch_jan04a/sketch_jan04a.ino
Using board 'uno2018' from platform in folder: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5
Using core 'arduino' from platform in folder: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5
Detecting libraries used...
/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216/sketch/sketch_jan04a.ino.cpp -o /dev/null
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
-> candidates: [SPI@1.0]
/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI/src /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216/sketch/sketch_jan04a.ino.cpp -o /dev/null
Alternatives for nRF24L01.h: [RF24@1.3.3]
ResolveLibrary(nRF24L01.h)
-> candidates: [RF24@1.3.3]
/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI/src -I/Users/clearlincognito/Documents/Arduino/libraries/RF24 /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216/sketch/sketch_jan04a.ino.cpp -o /dev/null
/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 -I/Users/clearli [... cut ...]
Used: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI
Multiple libraries were found for "nRF24L01.h"
Used: /Users/clearlincognito/Documents/Arduino/libraries/RF24
Using library SPI at version 1.0 in folder: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI
Using library RF24 at version 1.3.3 in folder: /Users/clearlincognito/Documents/Arduino/libraries/RF24
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.`_

Avamander, 2020-01-28:
Try adding this before any `#include`-s:
```c++
void digitalWrite(pin_size_t pin, bool value) {
    if (value) {
        digitalWrite(pin, HIGH);
    } else {
        digitalWrite(pin, LOW);
    }
}
```

It seems MegaAVR has some changes to digitalWrite making it a bit incompatible with the rest of the Arduino ecosystem.

clearlincognito, 2020-01-29:
I put that before my `#include RF24` and it gave me this error:
Arduino: 1.8.10 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

```
WARNING: library RF24 claims to run on avr, arm, x86, esp8266, esp32 architecture(s) and may be incompatible with your current board which runs on megaavr architecture(s).
/Users/clearlincognito/Documents/Arduino/libraries/RF24/RF24.cpp: In member function 'void RF24::csn(bool)':
/Users/clearlincognito/Documents/Arduino/libraries/RF24/RF24.cpp:69:27: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'
  digitalWrite(csn_pin,mode);
                           ^
/Users/clearlincognito/Documents/Arduino/libraries/RF24/RF24.cpp: In member function 'void RF24::ce(bool)':
/Users/clearlincognito/Documents/Arduino/libraries/RF24/RF24.cpp:80:51: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'
   if (ce_pin != csn_pin) digitalWrite(ce_pin,level);
                                                   ^
Multiple libraries were found for "nRF24L01.h"
 Used: /Users/clearlincognito/Documents/Arduino/libraries/RF24
 Not used: /Users/clearlincognito/Documents/Arduino/libraries/Mirf
Multiple libraries were found for "SPI.h"
 Used: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Avamander, 2020-01-29:
Yeah, I've investigated, requires more work to support because Arduino HAL devs made a backwards-incompatible change. 

Upstream issue: https://github.com/arduino/ArduinoCore-API/issues/25

Avamander, 2020-01-29:
Hmm, you can try this though, people in the thread say it can be fixed like this (before including again):

```c++
void digitalWrite(unsigned pin, bool value) {
    if (value) {
        digitalWrite(pin, HIGH);
    } else {
        digitalWrite(pin, LOW);
    }
}
```

clearlincognito, 2020-01-30:
I got this error when I tried that
`Arduino: 1.8.10 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

WARNING: library RF24 claims to run on avr, arm, x86, esp8266, esp32 architecture(s) and may be incompatible with your current board which runs on megaavr architecture(s).
/Users/clearlincognito/Downloads/NRF24-Tutorial/NRF24-receiver/NRF24-receiver.ino: In function 'void digitalWrite(unsigned int, bool)':
NRF24-receiver:2:41: error: call of overloaded 'digitalWrite(unsigned int&, PinStatus)' is ambiguous
     digitalWrite(pin, value ? HIGH : LOW);
                                         ^
In file included from /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/ArduinoAPI.h:52:0,
                 from /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/Arduino.h:23,
                 from sketch/NRF24-receiver.ino.cpp:1:
/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/Common.h:104:6: note: candidate: void digitalWrite(pin_size_t, PinStatus)
 void digitalWrite(pin_size_t pinNumber, PinStatus status);
      ^~~~~~~~~~~~
/Users/clearlincognito/Downloads/NRF24-Tutorial/NRF24-receiver/NRF24-receiver.ino:1:6: note: candidate: void digitalWrite(unsigned int, bool)
 void digitalWrite(unsigned pin, bool value) {
      ^~~~~~~~~~~~
Multiple libraries were found for "nRF24L01.h"
 Used: /Users/clearlincognito/Documents/Arduino/libraries/RF24
 Not used: /Users/clearlincognito/Documents/Arduino/libraries/Mirf
Multiple libraries were found for "SPI.h"
 Used: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI
exit status 1
call of overloaded 'digitalWrite(unsigned int&, PinStatus)' is ambiguous
`

Avamander, 2020-02-02:
```c++
void digitalWrite(unsigned pin, bool value) {
    digitalWrite(pin, (PinStatus)(value ? HIGH : LOW));
}
```

How about this, does that work?

clearlincognito, 2020-02-02:
Unfortunately, I got the same error
```

Arduino: 1.8.10 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

WARNING: library RF24 claims to run on avr, arm, x86, esp8266, esp32 architecture(s) and may be incompatible with your current board which runs on megaavr architecture(s).
/Users/clearlincognito/Downloads/NRF24-Tutorial/NRF24-receiver/NRF24-receiver.ino: In function 'void digitalWrite(unsigned int, bool)':
NRF24-receiver:2:54: error: call of overloaded 'digitalWrite(unsigned int&, PinStatus)' is ambiguous
     digitalWrite(pin, (PinStatus)(value ? HIGH : LOW));
                                                      ^
In file included from /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/ArduinoAPI.h:52:0,
                 from /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/Arduino.h:23,
                 from sketch/NRF24-receiver.ino.cpp:1:
/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/Common.h:104:6: note: candidate: void digitalWrite(pin_size_t, PinStatus)
 void digitalWrite(pin_size_t pinNumber, PinStatus status);
      ^~~~~~~~~~~~
/Users/clearlincognito/Downloads/NRF24-Tutorial/NRF24-receiver/NRF24-receiver.ino:1:6: note: candidate: void digitalWrite(unsigned int, bool)
 void digitalWrite(unsigned pin, bool value) {
      ^~~~~~~~~~~~
Multiple libraries were found for "SPI.h"
 Used: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI
Multiple libraries were found for "nRF24L01.h"
 Used: /Users/clearlincognito/Documents/Arduino/libraries/RF24
exit status 1
call of overloaded 'digitalWrite(unsigned int&, PinStatus)' is ambiguous
```

bxparks, 2020-02-11:
(Drive-by comment). You almost got it. Try using `uint8_t` instead of `unsigned`, like this:

```C++
#if ARDUINO_API_VERSION >= 10000
inline void digitalWrite(uint8_t pin, bool val) {
  digitalWrite(pin, val ? HIGH : LOW);
}
#endif
```
and insert this at the top of `RF24.h`.

(I tested that this compiles using Arduino 1.8.9 using UNO Wifi and Nano Every on various sketches under ./examples/).

The trouble with using `unsigned` is that it creates 2 versions of `digitalWrite()`:
```C++
void digitalWrite(unsigned, bool);
void digitalWrite(uint8_t, PinStatus);
```
Neither of them are an exact match for the call to `digitalWrite(pin, val ? HIGH : LOW)` (which is of type `digitalWrite(unsigned, PinStatus)`). So the compiler prints the error message.

Just for completeness, the following explicit cast will also work:
```C++
#if ARDUINO_API_VERSION >= 10000
inline void digitalWrite(uint8_t pin, bool val) {
  digitalWrite(pin, (PinStatus) val);
}
#endif
```
But this assumes that `PinStatus::LOW == 0` and `PinStatus::HIGH == 1`, which I cannot imagine someone would define differently. Then again, I never expected them to make a breaking change like this with a `PinStatus`.

As an aside, you might consider changing your `RF24::ce_pin` and `RF24::csn_pin` to be of type `uint8_t` instead of `uint16_t`. Every board that I am aware of uses `uint8_t` as the pin number for `digitalWrite()` and `pinMode()`. This also has the side effect of reducing the size of RF24 by 2 bytes.

As indicated in arduino/ArduinoCore-API#25, I agree with you that this breaking change was an incredibly poor idea. But it is what it is.

Avamander, 2020-02-11:
I added the overload to the library, would be nice if someone tested latest master out and saw if it works in their setup.

clearlincognito, 2020-02-15:
It works! Thank you @bxparks and @Avamander . I appreciate it greatly!

closed 2020-02-15 by clearlincognito