nRF24/RF24 #563: RF24 with 3 Pin

heavyeste, 2020-01-26:
Hello,
I try the library with 5 pin and the result is very good! With 10 meter and wall the signal is perfect. 
Now it's time to try with 3 pin configuration but i have many problems!
I followed the schema with capacitor and diode but the method 'isChipConnected' return always false!

I follow this schema and code:

https://www.instructables.com/id/NRF24L01-With-ATtiny85-3-Pins/#step1

Please give me support

regards
Esteban

Avamander, 2020-01-28:
If it returns false means there isn't a properly working connection. I unfortunately don't have an ATTiny85 to test out what exactly goes wrong here, assuming the Instructable author is correct you might want to check your setup (wiring, connections etc.) a few times. Alternatively try using a RF24 release before the Instructable was made, but AFAIK there shouldn't be any breaking changes to that, but you never know.

heavyeste, 2020-02-01:
I solved my problem! Mondate will arrive Attiny84! thank Amazon Prime! :)

closed 2020-02-01 by heavyeste