nRF24/RF24 #567: [Question] How to get RF24 library working with ATTiny84 ( SpenceKonde - ATTinyCore ) ?

3xtr3m3d, 2020-01-28:
Hi

I have wired NRF24L01 module as follow

```
ATTiny84-|- RF24
PA6-------|- MISO
PA5-------|- MOSI
PA4-------|- SCK
PA3-------|- CSN
PA2-------|- CE
```
and staring library like this

```
#define CE_PIN 2
#define CSN_PIN 3
#include "RF24.h"
RF24 radio(CE_PIN, CSN_PIN);
```

Using  SpenceKonde attinycore  RF module doesn't work. (tried clockwise and counter clockwise pinouts)  without changing any wiring, i tried arduino 1.6.4 with  damellis attiny core and RF module works.

is there anyway to get it working with this SpenceKonde attinycore? because lot of other stuff i want is missing in working core.

Regards

nopnop2002, 2020-02-03:
There are several Arduino Core Libraries for Tiny,
Not all ATtiny85 pinmaps are the same.

This library assumes the following pinmap:

	// ATMEL ATTINY84 / ARDUINO
	//
	//                           +-\/-+
	//                     VCC  1|    |14  GND
	//             (D  0)  PB0  2|    |13  PA0  (D 10)       AREF
	//             (D  1)  PB1  3|    |12  PA1  (D  9) 
	//             (D 11)  PB3  4|    |11  PA2  (D  8) 
	//  PWM  INT0  (D  2)  PB2  5|    |10  PA3  (D  7) 
	//  PWM        (D  3)  PA7  6|    |9   PA4  (D  6) 
	//  PWM        (D  4)  PA6  7|    |8   PA5  (D  5)        PWM
	//                           +----+

When using Arduino Core Library with different pinmap,
This can be done by changing pins_arduino.h.

nopnop2002, 2020-02-04:
This library has the following code for ATtiny.

	//ATTiny support code pulled in from https://github.com/jscrane/RF24
	#if defined(RF24_TINY)

	void SPIClass::begin() {
	    // set USCK and DO for output
	    // set DI for input
	        #if defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__)
	    DDRB |= (1 << PB2) | (1 << PB1);
	    DDRB &= ~(1 << PB0);
	        #elif defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
	    DDRA |= (1 << PA4) | (1 << PA5);
	    DDRA &= ~(1 << PA6);
	        #elif defined(__AVR_ATtiny2313__) || defined(__AVR_ATtiny4313__)
	    DDRB |= (1 << PB7) | (1 << PB6);
	    DDRB &= ~(1 << PB5);
	        #elif defined(__AVR_ATtiny861__)
	    DDRB |= (1 << PB2) | (1 << PB1);
	    DDRB &= ~(1 << PB0);
	        #endif // defined(__AVR_ATtiny861__)
	    USICR = _BV(USIWM0);
	}

	byte SPIClass::transfer(byte b)
	{
	    USIDR = b;
	    USISR = _BV(USIOIF);
	    do {
	        USICR = _BV(USIWM0) | _BV(USICS1) | _BV(USICLK) | _BV(USITC);
	    }
	    while ((USISR & _BV(USIOIF)) == 0);
	    return USIDR;
	}

	void SPIClass::end() {}
	void SPIClass::setDataMode(uint8_t mode){}
	void SPIClass::setBitOrder(uint8_t bitOrder){}
	void SPIClass::setClockDivider(uint8_t rate){}

	#endif

You can use the following libraries instead of this code:
https://github.com/JChristensen/tinySPI

3xtr3m3d, 2020-02-04:
> This library has the following code for ATtiny.
>
> ```
> //ATTiny support code pulled in from https://github.com/jscrane/RF24
> #if defined(RF24_TINY)
>
> void SPIClass::begin() {
>     // set USCK and DO for [...]

thanks i will check

2bndy5, 2020-10-27:
this repo now compiles with the SpenceKonde ATTiny core due to changes made by @jscrane in the recently merged #646 Stay tuned for the next release...

2bndy5, 2020-12-14:
Use v1.3.10 or later with the SpenceKonde ATTinyCore in Arduino IDE.

closed 2020-12-14 by 2bndy5