nRF24/RF24 #568: Arduino DUE don't work.

nopnop2002, 2020-01-28:
RF24 makes use of the extended SPI functionality available on the Arduino Due, and requires one of the defined hardware SS/CS pins to be designated in RF24 radio(ce_pin,cs_pin);

On the Arduino Due, the SAM3X has advanced SPI capabilities. 
It is possible to use these extended methods, or the AVR-based ones.
The extended API can use pins 4, 10, and 52 for CS.

But RF24.cpp don't use cs_pin.

My Code:

	/****************** User Config ***************************/
	/***      Set this radio as radio number 0 or 1         ***/
	bool radioNumber = 1;

	/* Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 7 & 8 */
	RF24 radio(7,10); // CE, CSN
	/**********************************************************/

Add debug print to RF24.cpp:

    #elif defined(XMEGA_D3)
    Serial.print("defined(XMEGA_D3) SPI.begin ce_pin=");
    Serial.println(ce_pin);
    if (ce_pin != csn_pin) {
        pinMode(ce_pin,OUTPUT);
    };
    Serial.print("defined(XMEGA_D3) SPI.begin csn_pin=");
    Serial.println(csn_pin);
    _SPI.begin(csn_pin);
    ce(LOW);
    csn(HIGH);
    delay(200);
    #else
    // Initialize pins
    if (ce_pin != csn_pin) {
        pinMode(ce_pin, OUTPUT);
    }

    #if !defined(LITTLEWIRE)
    if (ce_pin != csn_pin)
    #endif // !defined(LITTLEWIRE)
    {
        pinMode(csn_pin, OUTPUT);
    }

    Serial.println("SPI.begin");
    _SPI.begin();
    ce(LOW);
    csn(HIGH);
    #if defined(__ARDUINO_X86__)
    delay(100);
    #endif

Debug print:

	RF24/examples/GettingStarted
	*** PRESS 'T' to begin transmitting to the other node
	SPI.begin
	Sent response 0
	Sent response 0
	Sent response 0
	Sent response 0
	Sent response 0
	Sent response 0

TMRh20, 2020-01-30:
At one point rf24 used the extended SPI functionality of arduino due, but that was removed. You should be able to use just about any pins for ce & cs.

nopnop2002, 2020-02-02:
OK
Thank you.

closed 2020-02-02 by nopnop2002