nRF24/RF24 #571: Cannot set PA Power via Python3

jakimfett, 2020-02-09:
When calling `.setPALevel(RF24_PA_HIGH)`, it is expected that "PA Power" will be set to `PA_HIGH`.

However, the following python3 code returns `PA_MAX`, the default (see attached screenshot)
[image]
:

```
# An RF24 (specifically NRF24L01 w/ breakout adapter) receiver mockup for the Raspberry Pi Zero Wireless v1.1

# allows use of sys.exit()
import sys

# tracing per line
def trace(frame, event, arg):
    print("%s, %s:%d" % (event, frame.f_code.co_filename, frame.f_lineno))
    return trace

# Hardware interfacing, abbreviated.
import RPi.GPIO as io

# Use board numbering for GPIO stuffs, because portability
# see https://raspberrypi.stackexchange.com/a/12967
io.setmode(io.BOARD)

from RF24 import *

# enable stack tracing
sys.settrace(trace)

# use gpio pins 24 & 26 respectively (CE@26=0), and SPI speed of 32MHZ
radio = RF24(24, 0, 32)

# call .begin() first, or reap a segmentation fault
radio.begin()

print("Checking defaults...")
radio.printDetails()
print("Defaults printed.")

print("Setting PA level to ", RF24_PA_HIGH)
radio.setPALevel(RF24_PA_HIGH)

print("Printing details...")
radio.printDetails()
#radio.stopListening()
print("...exiting.")
sys.exit()
```
Using Raspbian Buster (`Linux rf24 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux`) on a Raspberry Pi Zero W v1.1

TMRh20, 2020-02-13:
Have you tested it otherwise? The 0xff results in printdetails indicates that SPI communication between the radio and RPi is not working. Also the max SPI speed for RF24 is 10mhz.

jakimfett, 2020-02-17:
>...tested it otherwise?

What're my options for testing this, aside from building an entire other Pi system and re-wiring the NRF to work with it?

---

> Also the max SPI speed for RF24 is 10mhz.  

Would that be fixed by using `radio = RF24(24, 0, 10)`?
What's the lowest I can set that, is 3 or 5 mhz too low?

TMRh20, 2020-02-17:
Well you can search for “RPi test SPI” to do a simple loop back test and make sure SPI is working.

From there you want to test SPI comms between radio/rpi. Double & triple check your wiring etc. until printDetails results shows a proper configuration. 
As an example, The pa + lna high power modules will require a separate power supply.

The constructor takes a regular integer, ie: 10000000 for 10mhz. Leave it blank for 10mhz.
See RPi and RF24 docs for available SPI speeds. 3 or 5 is fine.

jakimfett, 2020-02-26:
Thank you, that was exactly the information needed.

Will update once we've re-tested everything, hopefully yet this week...

closed 2020-03-29 by TMRh20