nRF24/RF24 #572: [Question] Why is SPI included in this lib?

bertmelis, 2020-02-19:
Why is there a "custom" SPI class included in this lib?

I'm using this lib on an ATtiny85 and in the Arduino framework (https://github.com/SpenceKonde/ATTinyCore) there is a lib available. Furthermore, when using other libraries that use SPI (like a BME280), there's no other way to adapt this library and use the framework's lib. Otherwise you get redefinition errors.

Avamander, 2020-02-19:
I didn't write the initial support for ATTiny so I don't know exactly, but I suspect SPI didn't properly exist in ATTiny board cores. If there's a way now to detect ATTinycore that has proper SPI then a preprocessor directive can probably be added. Pull requests are welcome.

jscrane, 2020-02-21:
Maybe something like this? (This works but is pretty nasty.)
```
diff --git a/utility/ATTiny/RF24_arch_config.h b/utility/ATTiny/RF24_arch_config.h
index dc5b0aa..635874f 100644
--- a/utility/ATTiny/RF24_arch_config.h
+++ b/utility/ATTiny/RF24_arch_config.h
@@ -9,6 +9,7 @@
 /*** USER DEFINES:  ***/
 //#define FAILURE_HANDLING
 //#define MINIMAL
+//#define CORE_HAS_SPI
 /**********************/
 
 #define rf24_max(a, b) (a>b?a:b)
@@ -24,9 +25,12 @@
 
 #include <stddef.h>
 
+#ifndef CORE_HAS_SPI
 // Include the header file for SPI functions ( Main SPI code is contained in RF24.cpp for simplicity )
 #include "spi.h"
-
+#else
+#include <SPI.h>
+#endif
 #define _SPI SPI
 
 #ifdef SERIAL_DEBUG
```

Avamander, 2020-02-21:
@jscrane 
It's not nasty if it's almost the only working way, I checked if `SPI_HAS_TRANSACTIONS` or `SPI_ATOMIC_FIX` could be used instead but the ATTinyCore has those defined even on versions where the SPI library doesn't even compile (according to git commit description).

Maybe ATTinyCore people could be nicely asked to define a version number somewhere, then RF24 could check for that?

TMRh20, 2020-08-19:
Lets admit, its kind of nasty. lol

Being the party responsible for adding the SPI code in for ATTiny, (stolen from @jscrane ) if I remember... I felt the need to chime in here. Anyway, I think its about time we just removed ATTiny SPI support from RF24 altogether. It was added in prior because there was not much for alternatives. 

Now that the ATTiny core has it included, I don't see any reason why we should maintain SPI support for ATTiny SPI here, probably just adds more confusion than anything. 

So, if this seems like a good idea and anybody gets a chance, I think the changes required are pretty straightforward, mostly just delete a bunch of stuff from RF24, and test with the examples etc to ensure functionality. I probably won't have a chance to do it for a while.

bertmelis, 2020-08-19:
I totally forgot about this issue. (because my ATTiny runs without any problem on a battery using SPI for 6+ months now).

Now, what I did to make my ATTiny code to compile was to comment out the `#include` to the SPI header in this lib. Otherwise it wouldn't compile because the core and this lib both define `SPI`. I've already looked into the SPI code itself and as far as I can see it's the same.

**I've only looked to ATTiny85.**

Avamander, 2020-09-19:
I added this to the [Hacktoberfest](https://hacktoberfest.digitalocean.com/) list of potential contributions.

The only thing expected is that someone removes the ATTiny-specific SPI workarounds from the library and tests on a **physical** device that it works with ATTiny core on at least two ATTiny's.

If someone has any other ideas, requirements or suggestions to this request, leave them below.

2bndy5, 2020-09-19:
I have been putting off wiring up 3V regulators, but I have a digispark board (ATTiny85) that I can test changes on. I'm assuming the 3-wire hack wouldn't be affected.

jscrane, 2020-09-19:
I have some [Tinysensors](https://github.com/jscrane/TinySensors) with ATtiny84 I can test on too if that helps...

closed 2020-10-11 by 2bndy5