nRF24/RF24 #573: Arduino uno + nrf

javierhouse, 2020-02-19:
Hello! I write because I have a doubt with the Arduino and nrf. I don't know, but if I execute the sketch with the Arduino uno, all works fine. I starts working with the example getting started, in the Arduino and in the pi3 and all works fine, then I change the sketch to do a loop and sends 5 times if I don't have reaponse, and all works fine and I send only 1 message, but if I extract the atmega328 of the Arduino board and put in the protoboard, all works fine, but I don't know way, all the time I have timeout in the response and I send the 5 times the same message. I have the same problem with the attiny 84 and 85. Is possible I need put something in the protoboard? Thanks in advance

Avamander, 2020-02-19:
If it works when directly attached but not on a protoboard you might be either having miswirings or getting additional noise. 

You can fix the noise on the power buses with electrolytic capacitors, you can google "decoupling capacitor" for wiring instructions.

javierhouse, 2020-02-20:
Hi! Thank you very much! I did a test this morning and when I put electrolytic capacitor in the Vcc and Grd of the nrf all start working fine! This evening I test wiht the attiny 84 and 85! Thank you very much, I had read something about this topic but I didn't understand correctly the problem... Now I know it jejejeje

javierhouse, 2020-02-25:
Hi everyone! I had tested the solution. If I put the electrolytic capacitor all work ok on the protoboard. Now I made the circuit in the PCB and the problem come again. I now the problem is the noise, because I put the nrf above of the attiny 84, but if I enlarge the connection with the typical protoboard cables the communication works perfect again. But in my case, the size is importan, what can I do? Do you have some tricks or recommendations? 
For now I change the PCB and works fine.

[image]

In the second one, is similar but don't work correctly y think the battery cables are the problem.

[image]

Thanks in advance!

javierhouse, 2020-02-25:
[image]

PhillyNJ, 2020-02-25:
I think the max voltage for the nRF is 3.3v and your batteries are pushing 3.6v. Not sure but I would try with a different power source. 

Regards,

Phil

Sent from Phil's iPhone

javierhouse, 2020-02-25:
Hello Phil. Thank you very much for your response. I changed the battery for the first circuit (works fine) to the second, and I don't have changes, the second doesn't works. I send correctly, but don't receive response. I changed the nrf too, and I nothing's changed.

Avamander, 2020-02-26:
So it doesn't work with thinner cables? Am I understanding you correctly? You could try an even bigger capacitor (maybe aim for lower leakage current ones if it's battery-powered).

If you had an oscilloscope or a signal analyzer you could take a look at the signal and power lines and see if there's a difference.

javierhouse, 2020-02-26:
Hello Avamander! No, I not talking about the size of the cables. Sorry mi English is... I need to improve :). I think the problem is the position of the nrf in the circuit. But with the same circuit, or very similar, one work ok an the

javierhouse, 2020-03-06:
Hi all! I test to change the capacitor to other. But the problem persist. :( I had review all the connections and everything are ok.

TMRh20, 2020-08-11:
Closing as this is a bit of an old issue and pretty much hardware related.

closed 2020-08-11 by TMRh20