nRF24/RF24 #576: [Question] Arduino Uno and Raspberry Pi aren't communicating with NRF24L01+

LowderPlay, 2020-03-08:
I have connected two identical NRF24L01 modules to Raspberry and Arduino, but they are not communicating. I have tried everything. At first I used NodeJs library on RPi, but now I'm trying with RF24.
This is my RPi code: 
```
#include <cstdlib>
#include <iostream>
#include <sstream>
#include <string>
#include <unistd.h>
#include <RF24/RF24.h>

using namespace std;
RF24 radio(24, 0);

int main(int argc, char** argv) {
    radio.begin();
    radio.enableDynamicPayloads();
    radio.enableAckPayload();
    radio.setChannel(0x76);
    radio.openWritingPipe(0xF0F0F0F0E1LL);
    radio.setPALevel(RF24_PA_LOW);

    // Dump the configuration of the rf unit for debugging
    radio.printDetails();

    while(true) {
        const char text[] = "Hello world!";
        cout << "sending " << text;
        if(radio.write(&text, sizeof(text)))
            cout << " done!" << endl;
        else
            cout << " fail!" << endl;
        delay(1000);
    }
}
```
And this is Arduino code:
```
#include <SPI.h>
#include "RF24.h"
#include "printf.h"

RF24 radio(9, 10);

void setup() {
  Serial.begin(115200);
  printf_begin();
  radio.begin();
  radio.enableDynamicPayloads();
  radio.enableAckPayload();
  radio.setChannel(0x76);
  radio.openReadingPipe(1, 0xF0F0F0F0E1LL);
  radio.setPALevel(RF24_PA_LOW);

  // Dump the configuration of the rf unit for debugging
  radio.printDetails();
  radio.startListening();
}

void loop() {
  while(!radio.available()) delay(10);
  Serial.println("receiving!");
  char packet[radio.getDynamicPayloadSize()];
  radio.read(&packet, radio.getDynamicPayloadSize());
  Serial.println(packet);
}
```

This is output from Arduino:
```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e31 0xf0f0f0f0e1
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e31
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x76
RF_SETUP	 = 0x03
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x3f 0x06
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_LOW
```
And from RPi:
```
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO24
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xf0f0f0f0e1 0x65646f4e31
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xf0f0f0f0e1
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x02
RF_CH            = 0x76
RF_SETUP         = 0x03
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x3f 0x06
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_LOW
sending Hello world! fail!
sending Hello world! fail!
sending Hello world! fail!
```
Maybe I'm doing something wrong?

TMRh20, 2020-03-08:
Have you tested with the provided examples? 

What type of modules are you using? Low power with/without antenna, high power pa+lna?

What are you using to connect the modules to the RPi? Usually, the shorter the wires the better.

LowderPlay, 2020-03-08:
Sorry, I've got my module working. Problem was on Arduino side, the module didn't send any data, maybe something wrong was with my connection.

closed 2020-03-08 by LowderPlay