nRF24/RF24 #577: Issue transmitting between two RPis

coopergooch, 2020-03-10:
Hello all,

I am running into an issue when i am trying to transmit between two RPis. I have both devices wired the same way, but sometimes when I run sudo ./pingpair_dyn on the second pi, the RX_ADDR_P0-1 values get flipped. When I run sudo ./pingpair_dyn on the first pi, I always get this:
`================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f0e1 0xf0f0f0f0d2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f0e1
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

 ************ Role Setup ***********
Choose a role: Enter 0 for receiver, 1 for transmitter (CTRL+C to exit) `
When I select 1 for the first pi to go into transmission mode, I get this and nothing on the receiver pi when it is set to 0:
`Now sending length 4...Failed, response timed out.
Now sending length 5...Failed, response timed out.
Now sending length 6...Failed, response timed out.
Now sending length 7...Failed, response timed out.
Now sending length 8...Failed, response timed out.
Now sending length 9...Failed, response timed out.
Now sending length 10...Failed, response timed out.
Now sending length 11...Failed, response timed out.
Now sending length 12...Failed, response timed out.
Now sending length 13...Failed, response timed out.
.
.
.`

When I run sudo ./pingpair_dyn on the second pi, I will either get the same as above or this:
`================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f0d2 0xf0f0f0f0e1
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xf0f0f0f0d2
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

 ************ Role Setup ***********
Choose a role: Enter 0 for receiver, 1 for transmitter (CTRL+C to exit) `

I once got the first pi to transmit a string in varying length, but I am now no longer able to. With either result of the second pi running sudo ./pingpair_dyn, I can't transmit. Does anyone have any idea as to what is happening? Any ideas or thoughts on the issue would be greatly appreciated.

TMRh20, 2020-03-10:
In the case of RX_ADDR_P0-1 addresses, the radio addresses are configured after printDetails(); is called, so this configuration will not be displayed initially. Since the RF24 module is not reset on RPi, it will retain the settings from previous runs. (displays correctly on subsequent runs)

In the case of things not working, standard troubleshooting applies. RPi can be picky sometimes.

What type of modules are you using?
What type of connectors are you using to the RPi? 
Double & triple check your wiring.
Are you able to test the hardware using Arduino?

coopergooch, 2020-03-10:
I am using two NRF24L01+PA+LNA models seen [here](https://www.amazon.com/Longruner-Transceiver-Antistatic-Compatible-LKY67/dp/B06WLH4ZG6/ref=asc_df_B06WLH4ZG6/?tag=hyprod-20&linkCode=df0&hvadid=344022943810&hvpos=&hvnetw=g&hvrand=7696381487183404444&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9032529&hvtargid=aud-801738734305:pla-623988174386&psc=1&tag=&ref=&adgrpid=69534739336&hvpone=&hvptwo=&hvadid=344022943810&hvpos=&hvnetw=g&hvrand=7696381487183404444&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9032529&hvtargid=aud-801738734305:pla-623988174386) connected with just some female to female wires connecting directly from RPi to nRF module which were purchased [here](https://www.amazon.com/gp/product/B07GD2BWPY/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1). Tonight, I will try to swap out some wires as well as check if adding a ~10u capacitor between Vcc and GND pins as I have seen that the nRF24L01+ models can have finicky power issues. 

Unfortunately I am unable to test with an Arduino at this point in time.

As far as RX_ADDR_P0-1, I have noticed that sometimes the values are reset at least upon RPi reset. The first time it is run after a restart, the TX and RX address values are 0xe7e7e7e7e7 and 0xc2c2c2c2c2 respectively and is updated on the second run. Is this what you mean by the configuration will not be displayed initially?

TMRh20, 2020-03-10:
Aha, your problem is likely related to your modules. The RPi cant supply adequate power, so you need a separate supply. I haven’t had much luck getting the pa+lna versions working with RPi myself, but have not done much testing. 

Yeah, the radio defaults is what you see there. The radios will retain whatever settings are applied until reset/power is removed. When running the sketch, the RX_ADDR_P0-1 configuration will change depending on which role is chosen, and the last configuration is what is displayed.

coopergooch, 2020-03-10:
Okay, I will go ahead and try to power it with a different power source. I will update with anything else that comes up. Thanks for all the help!

marcus011, 2020-03-19:
@coopergooch try setting Power to LOW or MIN
With radio.setPAlevel because raspberry 3.3v doesn't allow more current to draw on 3.3v rail

closed 2020-03-29 by TMRh20