nRF24/RF24 #578: pyRF24Network setup fails with "RF24Network.h: No such file or directory"

futureshocked, 2020-03-13:
Hi,

I am compiling the RF24 Python wrappers on a Raspberry Pi Zero.

I have done the exact same thing on a Raspberry Pi 4, with no issues.

On the Zero, I have increased the swap disk space to 1GByte, and after that it was quick to complite the RF24 Python wrapper with:

`$ python setup.py install`

**Here's the problem:**

I want to add support for RF24Network, so I do this:

```
$ cd ~/RF24/pyRF24/pyRF24Network
$ python setup.py install
```

Here is the output:

```
(lab_app) root@raspberrypi-zero:/var/www/lab_app/downloads/RF24/pyRF24/pyRF24Network# python setup.py install
running install
running build
running build_ext
building 'RF24Network' extension
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/var/www/lab_app/include -I/usr/local/opt/python-3.8.2/include/python3.8 -c pyRF24Network.cpp -o build/temp.linux-armv6l-3.8/pyRF24Network.o
pyRF24Network.cpp:3:10: fatal error: RF24Network/RF24Network.h: No such file or directory
 #include "RF24Network/RF24Network.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
error: command 'gcc' failed with exit status 1
```

**Additional context**
I am compiling using a Python virtual environment, with Python 3.8.2. 
The exact same process has worked on the Raspberry Pi 4.
I have installed all dependencies I am aware of:

```
sudo apt-get install build-essential
sudo apt-get install libncurses5-dev libncursesw5-dev libreadline6-dev
sudo apt-get install libbz2-dev libexpat1-dev liblzma-dev zlib1g-dev libsqlite3-dev libgdbm-dev tk8.5-dev
sudo apt-get install libffi-dev
sudo apt-get install libboost-python-dev
sudo apt-get install libpulse-dev
```
I have searched for the header file (RF24Network.h) but can't find it in the source code. Is this file generated?

How can I successfully compile the RF24Network Python wrapper for the Raspberry Pi Zero?

Thank you!

Peter

Avamander, 2020-03-13:
[RF24Network is a separate library.](https://github.com/nRF24/RF24/issues/578)

futureshocked, 2020-03-13:
Thank you for pointing me to the right direction, I got it to work. Will repeat a couple of time to make sure I got this.

closed 2020-03-13 by futureshocked